riscv/riscv-arch-test #1: Minor issues found

jstraus59, 2018-06-21:
A few minor issues found in the initial release:

1. The top level `README.md` has a reference to `doc/README.md` which does not exist (should it be `doc/README.adoc` instead?)

2. The setting of the `RUN_TARGET` macro in the  file `riscv-target/riscvOVPsim/device/rv32i/Makefile.include`  uses the path `$(ROOTDIR)/riscv-ovpsim/bin/Linux64/riscvOVPsim.exe` for the executable, which does not work on a Windows host (the `/Linux64/` needs be changed to `Windows64` for Windows hosts).

3. If I run with `RISCV_TARGET=spike` or `RISCV_TARGET=Codasip-simulator` and the target executable is  not on my path I do not get a useful error message.

4. Since there is a directory named `riscv-test-suite/rv32im` I tried running `make RISCV_TARGET=riscvOVPsim RISCV_DEVICE=rv32im`, but this does not work. This appears to be because the top level `Makefile` hard codes `ISA=rv32i`. However changing this still does not work because the directory `riscv-test-suite/rv32im` does not have a Makefile or other files that are in `riscv-test-suite/rv32i`.  It seems this directory should either be completed or deleted.

jeremybennett, 2018-06-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

> A few minor issues found in the initial release:
>
> 1.
>
> The top level |README.md| has a reference to |doc/README.md| which
> does not exist (should it be |doc/README.adoc| instead?)
>
> 2.
>
> The setting of [...]

Hi Jim,

Thanks for these. Do you want to submit a pull request for a corrected
version, or are you happy for me to just edit them in.

Best wishes,

Jeremy

> <https://github.com/riscv/riscv-compliance/issues/1>, or mute the
> thread 
- -- 
Tel:     +44 (1590) 610184
Cell:    +44 (7970) 676050
SkypeID: jeremybennett
Twitter: @jeremypbennett
Email:   jeremy.bennett@embecosm.com
Web:     www.embecosm.com
PGP key: 1024D/BEF58172FB4754E1 2009-03-20
-----BEGIN PGP SIGNATURE-----

iEYEARECAAYFAlsryuoACgkQvvWBcvtHVOFEfACfaZdKOlZp/wj8p4i8If9OX9Vw
07IAniK/BT/tr2IV/Tf3LOA2a64fDIiG
=dpY9
-----END PGP SIGNATURE-----

eroom1966, 2018-06-21:
Hi Jeremy

I will sort this issue
Thx
Lee

<https://github.com/riscv/riscv-compliance/issues/1#issuecomment-399154297>,
or mute the thread
<https://github.com/

eroom1966, 2018-06-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA1

> Hi Jeremy
> 
> I will sort this issue

OK - thanks. J

> Thx Lee
> 
>>> A few minor issues found in the initial release:
>> 
>>> 1.
>> 
>>> The top level |README.md| has a reference to |doc/README.md|
>>> which does not exist (should it be |doc/README.adoc| instead?)
>> 
>>> 2.
>> 
>>> The setting of the |RUN_TARGET| macro in the file 
>>> |riscv-target/riscvOVPsim/device/rv32i/Makefile.include| uses
>>> the path |$(ROOTDIR)/riscv-ovpsim/bin/Linux64/riscvOVPsim.exe|
>>> for the executable, which does not work on a Windows host (the
>>> |/Linux64/| needs be changed to |Windows64| for Windows
>>> hosts).
>> 
>>> 3.
>> 
>>> If I run with |RISCV_TARGET=spike| or 
>>> |RISCV_TARGET=Codasip-simulator| and the target executable is
>>> not on my path I do not get a useful error message.
>> 
>>> 4.
>> 
>>> Since there is a directory named |riscv-test-suite/rv32im| I
>>> tried running |make RISCV_TARGET=riscvOVPsim
>>> RISCV_DEVICE=rv32im|, but this does not work. This appears to
>>> be because the top level |Makefile| hard codes |ISA=rv32i|.
>>> However changing this still does not work because the directory
>>> |riscv-test-suite/rv32im| does not have a Makefile or other
>>> files that are in |riscv-test-suite/rv32i|. It seems this
>>> directory should either be completed or deleted.
>> 
>> Hi Jim,
>> 
>> Thanks for these. Do you want to submit a pull request for a
>> corrected version, or are you happy for me to just edit them in.
>> 
>> Best wishes,
>> 
>> 
>> Jeremy
>> 
>>> <https://github.com/riscv/riscv-compliance/issues/1>, or mute
>>> the thread
>>> 
>>
>>
>>
>> 
> thread.
or mute the thread
- -- 
Tel:     +44 (1590) 610184
Cell:    +44 (7970) 676050
SkypeID: jeremybennett
Twitter: @jeremypbennett
Email:   jeremy.bennett@embecosm.com
Web:     www.embecosm.com
PGP key: 1024D/BEF58172FB4754E1 2009-03-20
-----BEGIN PGP SIGNATURE-----

iEUEARECAAYFAlsr0DkACgkQvvWBcvtHVOG/ggCXdKedjZXJTAZZCmzcwQofIx3D
pQCfT5CsTyyn204413u8kU3OTYFtJjM=
=VTIi
-----END PGP SIGNATURE-----

jeremybennett, 2018-06-22:
Thanks for the fix from Jim. Fixed in commit [5c47067](https://github.com/riscv/riscv-compliance/commit/5c4706719a1e6d2916c4960c8ee370f9449b16fe)

closed 2018-06-22 by jeremybennett