riscv/riscv-arch-test #3: 2.4 Processor configuration clarification

kasanovic, 2018-07-03:
Section 2.4 on processor configuration is unclear.  There are two orthogonal issues.  One is that a configurable implementation has to have its options specified before it can be used.  The second is that the compliance test framework has to be told what features to test.  The compliance test should only cover a subset of the configuration's supported features.  This implementation configuration belongs in the target environment section, as it is part of selecting the target to be tested. This section 2.4 should be about the compliance test configuration, indicating how to instruct it to test a certain set of features.

allenjbaum, 2019-12-10:
I think this has been addressed in Pull#65, which is the "next generation" test format spec. This is currently undergoing review and is intended to match the riscof framework. There is a DUT spec, and a Target spec. The intent is to test all features in the DUT that are ]required for the target (e.g.platform). The "default" case is that the target is simply RV32i, so no options required.

neelgala, 2020-03-19:
@allenjbaum since the test-format-spec has been frozen can we close this?

allenjbaum, 2020-03-19:
Yes,  all these issues have been addressed in testformatspec v1.2.5

allenjbaum, 2020-03-19:
So, effectively fixed by pull#65

closed 2020-03-21 by neelgala