riscv/riscv-arch-test #4: Section 2.3 Target Environment

kasanovic, 2018-07-03:
The target environment section should explain that software is likely a component of the target environment.   I would like to use the phrase "target test execution environment".  The "execution environment" for a RISC-V system is something that is actually defined in Section 1.2 of the user ISA manual.  It would be good to build on that here.

allenjbaum, 2019-12-10:
I am unclear which sec 2.3 this is referring to; I don't the this is the testformat spec.
Bu think this has been updated in pull65; the relevant sentence is :Selection of tests is performed based on the target’s asserted configuration, and the spec, Execution Environment or platform requirements.

allenjbaum, 2020-02-25:
Krste: please look at rev1.2.5 of the TestFormat spec and see if it fixes your documentation (I think) issue: https://lists.riscv.org/g/tech-compliance/files/Review%20Documents

neelgala, 2020-03-25:
updates on this?

kasanovic, 2020-03-30:
I'm OK with general structure of this.  The TestFormat doc obviously needs some work to make it more readable.  I think it is important to spell out that the target shell is the responsibility of the vendor, though there may be some standard shells available from which to start.  The list of test target types is OK but not exhaustive and should probably be listed as "includes" rather than implying "is only" one of these types.

closed 2022-08-11 by neelgala