riscv/riscv-arch-test #5: rv32imc C.LWSP.S test5 is read a word outside the binary

Dolu1990, 2018-10-11:
https://github.com/riscv/riscv-compliance/blob/master/riscv-test-suite/rv32imc/src/C.LWSP.S#L86
test_data + 252 > .word 31
It is currently passing simulators tests because they have their memory cleared before starting the test (probably).

Imperas, 2018-10-18:
Great catch, thanks for the heads up.

You are right that a simulator has a default setting for memory values (in this case all 0's)
in the Imperas simulator we can also pattern an alternate memory value, a quick change of the default to **0xdbdbdbdb**
and this highlighted what you saw on hardware
The test was setup incorrectly, we have committed a fix I also recoded C-LW.S to use the same idiom for the test_data layout as C.LWSP.S
- please let us know if this reproduces correctly on hardware.

Imperas, 2018-10-18:
clsoe #5

Imperas, 2018-10-18:
close #5

Dolu1990, 2018-10-22:
Currently the flow which pointed me the bug isn't stable, i will try this test again :)
Thanks

closed 2018-10-22 by Dolu1990