riscv/riscv-arch-test #9: Setting SATP and PMP should be optional

neelgala, 2018-10-22:
Line [145](https://github.com/riscv/riscv-compliance/blob/fa42764b059db95dad1563bf20aa1a83351da080/riscv-test-env/p/riscv_test.h#L145) and [146](https://github.com/riscv/riscv-compliance/blob/fa42764b059db95dad1563bf20aa1a83351da080/riscv-test-env/p/riscv_test.h#L146)  in the riscv-test-env/p/riscv_test.h should be optional and not mandatory. 

The mplementations not supporting Supervisor mode will not have satp register and thus would trap again. Probably the macro RVTEST_ENABLE_SUPERVISOR can be used for this.

Similarly, setting/reading PMP should be optional based on user defined inputs - not sure if there is a macro to do this.

aswaterman, 2018-10-22:
They are optional. The code executes correctly on processors that do not have those CSRs.

neelgala, 2018-10-22:
That is because you are changing mtvec in the INIT_PMP and INIT_SPTBR macros. RISC-V allows you have a read-only mtvec (page 26 of the latest draft), in which case this will fail - wouldn't it?

aswaterman, 2018-10-22:
Yes, that’s true, those macros assume writable mtvec.

allenjbaum, 2019-12-14:
The riscof framework will enable tests to list required extensions, options, and WARL field illegal to legal mappings and skip tests that don't meet them. This is another test that will need to be rewritten to use this when riscof becomes ratified.

closed 2020-02-26 by neelgala