riscv/riscv-arch-test #11: illegal.S in rv32mi generates supervisor interrupt - might not be supported on all implementations

neelgala, 2018-10-27:
Line [39](https://github.com/riscv/riscv-compliance/blob/ff7ecdb76b15dad11d8b17566cce9d28f72e38be/riscv-test-suite/rv32mi/rv64mi/illegal.S#L39) if illegal.S sets the SSIP bit. Implementations not supporting supervisor mode can choose to not have this bit, and thus such implementation will get stuck in the self-loop that follows.

Solution: user can provide  the address or macros of the memory-mapped control registers responsible for setting MSIP as part of the compliance_test.h. These can be inserted instead of line 39 and expect the same effect then on.

allenjbaum, 2018-10-31:
In theory, these tests are specifically intended to check the unprivileged ISA and to not depend on ANY privileged op or resource

There should be a parameter passed to framework that indicates whether compressed is supported or not, and based on that the environment should include or skip tests that either check or use compressed ops (and halfword aligned ops). One reason for this is that tests that check if compressed support is present require access of MISA CSR (or an ECALL that returns the answer)

Both of these require execution of a privileged op or access of a privileged CSR, and these will be tested by the privileged ISA test suite(s) - which may be the same as the unprivileged suites but also check for differences between modes, and even emulation of ops that get illegal instruction traps.

neelgala, 2018-11-04:
@allenjbaum 
Yes these are not "privileged" tests per se, but these tests need to have a mechanism to probe the underlying implementation specs and then decide which test-cases to execute and which to bypass. Some mandatory CSRs like mstatus can be used to carry out some limited probing. The fact that MISA is an optional CSR limits detection of all supported ISA extensions on an implementation.

I don't think we can avoid using CSRs for "unprivileged" tests but we should probably aim at reducing their usage.

neelgala, 2018-11-04:
For the illegal.S test I realized that this [snippet](https://github.com/riscv/riscv-compliance/blob/0d19ad8646de872913f943f561877d4241495bff/riscv-test-suite/rv32mi/rv64mi/illegal.S#L28-L35) of code checks if supervisor is supported or not. If it is not supported then the program simply exits with a different signature than the one in the reference folder.

Also some of the tests in this file are not specific to supervisor like the wfi and vectored interrupts, which can be easily modified to work with machine/machine+user supported implementations.

Following is what I can see moving forward.

1. Multiple-signatures:  We can either maintain multiple reference signatures for each such test and expect that an implementation matches atleast one of them to declare a "pass". Now this behaviour can be found in other tests also. I am not sure if the number of reference signatures per test will exceed 2 or not.
2. Probing and Bypass: Each test-case/test-file should have a prologue section which can probe the implementation to check which test-cases in the file should be executed and thereby make decision on how to create a single-signature dump.
3. Self-checking: since some test-cases are not executed on some platforms each test-case can be equipped with having to check its own signature rather than dumping out the entire signature out. This would avoid having to maintain multiple reference signatures for a single test file.

allenjbaum, 2018-11-05:
> Found my mistake at my end. The signatures whether supervisor is enabled
> or disabled will be the same. But I still hold the rest of the arguments in
> the earlier comment.
> Another use-case with different signatures is ma_fetch in rv32mi when run
> with and without compressed enabled.

This issue has been discussed (primarily by me) and answered and documented
in the compliance TG doc site (somewhere).

Unprivileged ISA tests should not fault; faulting is a property of a system
with privileged ops.
That means that unprivileged ISA tests will not test whether you are in a
privileged mode, because that requires a privileged op (CSRRW of a Mmode or
Smode CSR).
Privileged ISA tests will be able to test what mode a hart is in, and will
check that CSRRW of MISA will trap when the hart is in Umode, for example.
Note that it will not test that reads to Smode CSRs trap if executed in
Umode - or even in Smode if Smode does not exist.
I would expect that any op that can be executed in Umode, but behaves
differently in different modes (primarily Load,Store,Branch) will have
duplicate tests that checks to see that their behavior really is different
in the different modes.

Basically, the compliance framework should not run a test that require
device resources that the framework has not been told that the device
supports.
It is not the responsibility of any test to check to see if a resource
exists - the test framework should just know.
It is valid for a test to check if a resource exists if it has been told
that it does.

So, a single signature per test, but different tests depending on
parameters passed into the framework (e.g. precisely which options are
implemented)
(an alternative is to actually pass parameters to the test compilation, so
a single test source would generate but multiple binaries, depending on the
implementation)

hajekra, 2018-12-10:
Hi Neel,

when I run illegal.S test with OVPsim simulator, any configuration with USER mode enabled fails (timeout).

I added following parameter to run the target with USER mode enabled:
`--override riscvOVPsim/cpu/misa_Extensions=0x100100`

Problem seems to be in the snippet you have posted. "Vector interrupt" part of the test should be skipped without S mode but it isn't in this case. Is this what you have reported? And is it caused by the test or by the simulator?

jstraus59, 2018-12-13:
@neelgala 

You are correct that there is an issue in the riscvOVPsim. The mstatus.xPP fields are defined in section 3.1.7 of the [Privileged ISA Specification v1.10](https://riscv.org/specifications/privileged-isa/) as _Write Legal Read Legal_ (WLRL) . The definition of a WLRL field states "_Software ... should not assume a read will return a legal value unless the last write was of a legal value._" This means that under the 1.10 spec you cannot assume that writing an illegal value to the MPP field will return a different value on a subsequent read, as this code snippet does. 

When an illegal value is written to mstatus.MPP, the riscvOVPsim model was returning the same value that was written, which is allowed under the 1.10 architecture spec, but which means the code snippet does not work as intended.

However in the latest [draft of Privileged ISA Specification v1.11](https://github.com/riscv/riscv-isa-manual/releases) the mstatus.xPP fields have been changed to be _Write Any Read Legal_ (WARL). This means that after writing an illegal value to the MPP field a legal value must be returned. Since this is also an allowed behavior under the 1.10 spec we have updated riscvOVPsim to simply ignore writes of illegal values, leaving the field unchanged. This behavior makes the code snippet behave as expected.

Pull request #26 includes this fix. Thanks for pointing this out.

Jim Straus

P.S. Note that the MPP value for Supervisor mode is actually  `2'b01`, not the `2'b10` value you mentioned, which is the value the code snippet writes.. Also the code snippet checks to see if the value that is read is the same one it wrote, not that it is  `2'b11` as you indicated.

hajekra, 2018-12-14:
@neelgala 
Thank you for your explanation. I didn't analyse it since I saw it is already reported.

@jstraus59 @eroom1966 
I reran all compliance tests with the riscvOVPsim with all valid configurations and it seems to be fixed.

closed 2020-02-26 by neelgala