riscv/riscv-arch-test #13: compressed tests are using non-compressed instructions

hajekra, 2018-11-02:
Following macros in test_macros.h are used for testing compressed instructions but do not use it directly:
TEST_CLWSP, TEST_CSWSP, TEST_CADDI16SP, TEST_CADDI4SPN.

eroom1966, 2018-11-02:
Hi hajekra

As I recall, I could not get the assembler we were using to accept some of the compressed instructions by name, for example c.addi16sp

I have found that constructing an assembler instruction correctly, it will map to the compressed instruction
are you saying that you have been able to assemble code using the mnemonic 
c.addi16sp ?

If so, can you tell me what version of the assembler you used ?
Thx

hajekra, 2018-11-02:
Hi,

I am using riscv64-unknown-elf-gcc (GCC) 7.1.1 20170509. In asm file I used "c.addi16sp sp, 0x20".

Radek H.

Imperas, 2018-11-02:
@hajekra 
Thanks for clarification
fixed in commit 57049ee

hajekra, 2018-11-07:
@Imperas 
Thanks, closing this issue.

closed 2018-11-07 by hajekra