riscv/riscv-arch-test #16: I-MISALIGN_JMP-01.S assumes compressed can be turned off

neelgala, 2018-11-07:
Line [51](https://github.com/riscv/riscv-compliance/blob/0d19ad8646de872913f943f561877d4241495bff/riscv-test-suite/rv32i/src/I-MISALIGN_JMP-01.S#L51) tries to switch off the compressed support before proceeding with the tests.
Implementations with either of the following specs:

- do not implement MISA
- implement MISA but "C" bit is readonly and hardwired to "1" 

will fail to execute the test completely.

Possible solutions:

- re-arrange instructions and trap-handler such that an implementation with compressed support traps with illegal instruction, updates signature and jumps back to next test. Similar to what is done in ma_fetch.

- Have compile time options to generate a different elf for different target devices.

hajekra, 2018-11-08:
Why does it fail? MISA should be WARL register. If MISA is hardwired or isn't implemented, test should continue anyway. It will fail only in case C is implemented because trap is not executed. So I would say implementations with C extension (without capability of turning C off) will fail.

I agree that writing to MISA should be removed. At the same time test should be skipped for configurations with C extension. But there is one other issue in this test (and MISALIGN_LDST too) - trap handler, because spec supports hardwired mtvec.

neelgala, 2018-11-09:
Yes you are write "implementations with C extension (without capability of turning C off) will fail".
That is the case I am referring to. Probably couldn't convey it right in the description. 

If you skip a test you will still have to manage generating the signature which matches the reference signature. Since the signature is updated at multiple places: within the test-cases themselves and in the trap handler.. I am finding it hard to figure out a clean fix.

The issue of considering mtvec to be readonly is captured in issue #9 .

aprnath, 2018-12-10:
Turning off compression could be made conditional?

aprnath, 2018-12-10:
Actually, the RISCV way would be to check (via a write-0-read) to see if "C" can be cleared. If it can be cleared, proceed with the clearing and the test. If "C" cannot be cleared, test should provide alternal paths to success.

aswaterman, 2018-12-10:
Two reminders:

- Make sure the PC is 4-byte aligned when clearing misa.C, or the write will be suppressed.

- Clearing misa.C does not necessarily guarantee that IALIGN increases to 32 bits. There could be other extensions that maintain IALIGN of 16 bits, e.g. 48-bit extensions.

RolfyYu, 2019-01-21:
I think the tests may have 3 results:
PASS/FAIL/SKIP.
like @aprnath mentioned,
If the C-bit can be turn off correctly,  test should keep running.
If C-bit is still on after trying to clear it, test should end and leave a result of SKIP.

allenjbaum, 2019-01-30:
As I've commented in several other issues: tests that use privileged resources should not be in the unprivileged test suite. MISA is inherently privileged; QED it should be in the privileged test suite (which doesn't really exist yet).

To another point: WARL CSRs are a pain for compliance - but unimplemented CSRs are a special case, and tests that depend on CSRs than can be unimplemented must compensate for that.
That may require tests to have multiple signatures (which is going to be a pain) or may require that a list of unimplemented CSRs must be passed to the framework, which either modifies the test or selects a different test (in which case there are effectively 2 tests but each with a unique signature).

allenjbaum, 2019-12-14:
The riscof framework uses a YAML file to describe extensions and options implemented by a device (and the illegal to legal value mapping for WARL fields) and employs a macro in each test case that contains the required options to run it. This test will eventually be re-written to use that information, and the test will be skipped if MISA.C can't be cleared.

neelgala, 2020-03-19:
closing this as it has been fixed in RISCOF and cannot be fixed in the current framework without major revamps

closed 2020-03-19 by neelgala