riscv/riscv-arch-test #22: I-MISALIGN_LDST-01 assumes misaligned data access will trap

brouhaha, 2018-11-25:
I-MISALIGN_LDST-01 assumes misaligned data access will trap. Misaligned data access traps are NOT required by the ISA spec v2.2 nor the privileged spec; it is entirely reasonable for the misaligned access to be handled in hardware.
The test should be written to succeed if either the misaligned access completes correctly with no trap, or traps with the appropriate values in the CSRs.
This issue came up when I was testing a core I designed for the RISC-V soft CPU core design contest, which requires that the core pass all 55 RV32I tests. My core implements misaligned data access in hardware, but I had to add extra stuff to my core to implement the misaligned data access traps to pass the test.

neelgala, 2018-11-26:
@brouhaha 
Yes you are right. This test apart from checking misaligned data-access behavior also checks trap mechanisms. And as you pointed out traps on mis-aligned data-access are not mandatory. I think this test would be better suited for "privilege tests"

I have added a new test to check misaligned data-accesses. You can see them in [this](https://github.com/riscv/riscv-compliance/pull/23) pull-request. 
Would be helpful if you could confirm that you get the same signature for this test using your HW.

Thanks,

hajekra, 2018-11-26:
Current implemetation of I-MISALIGN_LDST-01 is testing misalign trap and it should be skipped if misalign memory access is supported in HW. It is same case as I-MISALING_JMP-01 which is valid only for implementations without compressed extension.

@neelgala 
In my opinion, the better approach is to add new test for HW support while keeping the original one for traps. Framework should select which one to run for given configuration. I think we should keep tests simple and we shouldn't mix features together.

The reasons why I think it's better:
1. code maintenance - when someone will change one part of the test (for example adding extra check for MEPC and others CSR registers etc.) he won't break the second part of the test unintentionally (for example when he adds extra signature output to trap handler). Moreover right now there is no way how to prove the HW support part of the test is correct. As far as I know spike does not support this.

1. there is possibility that GM and tested implementation will execute different code of the test which makes it very hard to debug. I think in general GM should always run same code as tested implementation and we should avoid ambiguous program flow. 

1. makes both tests easily extensible in the future.

jstraus59, 2018-11-26:
Hi Radek
your comment

'Framework should select which one to run for given configuration. I think
we should keep tests simple and we shouldn't mix features together'

yes it does - today in a very simple way, where you select the suite... for
example

make RISCV_ISA=rv32i

    this runs RV32 I instructions in machine mode

but it needs much further discussion as there is the other school of
thought where the test explores what is implemented

there are many many configuration options and this means there would need
to be a test for each various cross multiplied configuration set meaning an
explosion in the number of tests and then a framework to select them all
etc the framework then becomes very complex - the effort moves to the
framework and not the tests
or the other approach is a few more complex tests

hence more discussion is needed

it is like the choice of self checking tests or a signature file to compare
with - there are pros and cons of each option

and probably what needs to happen is to carefully document the different
approaches in detailed proposals and then have the community of compliance
concerned companies make a decision - this was done for the choice of
signature or self checking and the decision was signature - which is the
current place/direction we are at.

I personally am in favour of smarter tests that explore the DUT (which can
be done in the mandatory machine mode) and then exercise and check what is
enabled in that device for a specific suite

in this test specific instance (mis-aligned load/store) there is detailed
discussion needed about what is mandatory in hardware and what is optional
etc...

all for future discussions...

Simon

-- 
====================================================================
The information contained in this electronic mail message and any
attachments hereto
is privileged and confidential information intended only for the use of the
individual or
entity named above or their designee.  If the reader of this message is not
the intended
recipient, you are hereby notified that any dissemination, distribution or
copying of this
communication is strictly prohibited.  If you have received this
communication in error
please immediately notify us by return  message or by telephone and delete
the
original message from your mail system.  Thank you.
====================================================================

allenjbaum, 2019-01-30:
I've commented on this before, but I do it again here.

Misaligned support is one of the few implementation options that is defined in the unprivileged RV architecture spec. It has always  been the intention that misaligned support should be parameter passed into the framework. The framework should ensure that, for the unprivileged test suite, no test is performed that requires privileged resources.

 A misaligned trap must use privileged resources, so they can't be allowed in the unprivileged test suite.

This can be handled by 
 - modifying the test so that it checks whether or not misaligned is supported, and avoiding misaligned loads (either by branching around them or forcing address alignment
 - not choosing tests that have misaligned operands (and ensuring they only get executed by privileged mode test suites).

Since there is no architectural mechanism that allows querying whether misaligned accesses are supported in hW, the second option should be used.

allenjbaum, 2019-12-14:
Update: the riscof framework will have a YAML cofiguration file that describes what the implementation supports. This includes the  extensions defined in the configuration string, and many other things (how CSR WARL fields respond to illegal values, which optional CSRs are implemented, and whether unaligned accesses are supported. This will be parsed, and used to select which tests will be run. There will be two different tests: one that expects HW supported unaligned accesses, and one that expects traps (alternatively, we could have a test that returns 2 different signatures, but the reference model will still have to be configured to support unaligned access or trap in any case)

allenjbaum, 2022-07-12:
The latest framework release, riscof, should have fixed this issue.
Could you confirm that this is or is not the case so we can close this?
Note that in order to do this you'll need to install and configure the new framework, 
and create the YAML file that describes your hart (and specifically, misalign support or lack of it)

allenjbaum, 2022-08-11:
Needs to be coordinated with Sail (& Spike). Should be closed when Sail parsing of YAML file is conplete

allenjbaum, 2023-02-22:
It is possible to configure Sail to support or not support misaligned load/store, and the test infrastrucutre should handle this correctly now. It may be necessary to manually add the misaligned support to the Sail command line with -m to force it to be configured for misaligned support.

closed 2023-02-22 by allenjbaum