riscv/riscv-arch-test #27: Macros are checking side-effects

jlucnagel, 2018-12-21:
The macros are target dependent and are implemented in compliance_io.h, e.g. RVTEST_IO_ASSERT_GPR_EQ. Some registers are said to be reserved and are used by the tests.

False errors get reported by macros because the checked value is a side effect of the macro implementation. As an example rv32i/src/I-ADD-01.S line 84: the code asserts x5 is 0, but the signature checks it should be 1.

The macros are pretty useful because they basically run in many environments. The asserted value should check the real value, and not a side effect of the checking macro. If a macro is using reserved registers, then it should check it and not execute it if a conflict occurs. It is likely possible to use target specific macros that can use fixed memory locations to swap registers and thus avoid solve reserved register issues.

Last comment: by not implementing the macros, it is possible to skip it. If macros are choosen, it is however not possible to skip saving the signature to memory. It would probably save memory and accelerate tests if it was possible to disable signature saving when using self-checking macros.

eroom1966, 2019-01-23:
I think the usage of the reserved registers as an argument to the ASSERT statement is incompatible, and should not be used.

jlucnagel, 2019-01-24:
We modified the assert macro to swap registers if the register to test is a "reserved register". But our change is target dependent (currently uses address 0 of the memory map). Skipping the comparison with a side effect register within the macro would be another solution that would reduce the coverage but that would not break the test for those who have a target dependent way to run it.

allenjbaum, 2019-01-24:
note that if the memory map passed to the framework doesn't implement
address 0 (pretty common to make sure it doesn't exist to catch null
pointers), then this test can't be run.

hajekra, 2019-01-24:
@jlucnagel 
What about using "signature_end" memory location? This might be safe for most of the platforms.

> But our change is target dependent (currently uses address 0 of the memory map).

I don't think it is a problem - all these IO macros are target dependent.

In my opinion, we should avoid "reserving" registers. For ASSERT macro we could add one extra input register, which would be used for loading immediate and comparison. It is quite safe to use the register which was used as address to signature - it is not needed anymore and it cannot be checked by ASSERT.

Problem is in WRITE_STR macro, which needs more than 1 register and the number of registers may be also target dependent. On the other hand, if it is used only before and after small independent tests, it should be safe to use any register there...

> Skipping the comparison with a side effect register within the macro would be another solution that would reduce the coverage

What do you mean by this? I don't think that coverage should be measured with these macros on. Skipping ASSERT and other IO macros should not break the result coverage.

eroom1966, 2019-01-24:
I think the best solution is to perform a register save/restore to a defined memory area around any of the assertions, log writes etc. this would at least reduce the number of reserved registers to 1, which would be the register used in the register STORE/LOAD operations

allenjbaum, 2019-01-24:
The issue with address 0 ( indeed anything in the first 2k) is that unprivileged tests that are checking that loads/stores/atomic using X0 as a base can’t be run unless that address range has been passed to the framework as legal. Privileged mode test suite tests can do this, since they can map VA0 to someplace legal

-Allen

allenjbaum, 2019-12-10:
HAs this been addressed in the relevant macros by adding a scratch register to their definition? The scratch register is used as a temporary to load a save area address so registers can be saved and restored. That scratch register cannot be saved by any signature store or checked with an assert. We should define a standard save area label for this if this hasn't already been done.

allenjbaum, 2020-04-22:
@jlucnagel : please close this if the problem has been fixed, or show that it is still broken

closed 2020-05-14 by jlucnagel