riscv/riscv-arch-test #30: The golden results of rv32ui and rv64ui should be different

RolfyYu, 2019-01-21:
Since there is a "CHECK_XLEN" macro, 
the first word(double word) of begin_signature will set to 1 if XLEN is 64.
If XLEN is 32, the first and second word of begin_signature are 0xffff_ffff.

eroom1966, 2019-01-23:
At the moment these were setup as placeholders, the rv64* all need setting up correctly, and enhancing the coverage.

allenjbaum, 2020-02-25:
Is this a nice-to-have, or does cause tests to fail?
I know some tests that use constants fail in RV64 because constants are not sign extended in RV64, but that is slightly different than what I think this is about.

neelgala, 2020-02-27:
the rv64* tests would require a focused effort. As Lee mentioned, they have to re-written with coverage-driven-strategy rather than re-use-rv32 strategy. Since this is a good enough effort I would suggest diong this in the next version.

allenjbaum, 2020-02-28:
I tend to agree, but it means we need to start moving to approve the next
version fairly quickly - which also means moving legacy priv-mode tests.
That's even more work than just updating RV64i tests.
We should do some triage and figure out an order of transitioning the tests.

PR #101 (2020-03-25): Restructure
Changes in this pull-request:

    * restructuring the riscv-test-suite to indicate clearly what is deprecated, wip and usable
      tests.
    * based on the above fixed the directory structure for riscv-targets where-ever applicable. Only
      tested riscvOVPsim and spike.
    * fixed script bugs for spike as well
    * renamed rv32i/I-IO.S to rv32i/I-IO-01.S along with necessary changes to the reference files
      and Makefrag
    * renamed mbadaddr csr to mtval as raised in issue #31
    * C.SWSP-01.S test updated to fix issue #37

Close: #8 
Close: #31 
Close: #30 
Close: #3 [... cut]
closed 2020-03-25 by neelgala