riscv/riscv-arch-test #37: rv32imc C.SWSP test5 writes a word outside the binary

astimonov, 2019-01-31:
Related issue: #5 
https://github.com/riscv/riscv-compliance/blob/d8e99ee55d5fa3460c0a2f0898dd917a616763fb/riscv-test-suite/rv32imc/src/C-SWSP.S#L86

The test performs a write to offset 252, which is outside of section test_data (32 words).

allenjbaum, 2019-12-10:
This does need to be fixed; if we are only testing signatures within the test_data section, then this test is effectively bypassed, since its signature isn't checked

eroom1966, 2019-12-10:
This is a good catch, and more importantly indicates the necessity of the mutation fault testing to be completed for all suites - up until now we only checked this for the RV32I tests

allenjbaum, 2019-12-11:
If we start to use the SIGUPD macro, I don't think this can happen in
general, though this test deliberately stored values in non-contiguous
locations (don't quite understand why). This test could to use a SIGBASE
macro to set the base to a specific value, which would catch this error
In general, SIGUPD and SIGBASE should do bounds checking and give an
assembler error if basereg+offset is outside the bounds of DATA_START and
DATA_END, and that would catch this type of error.
I am proposing a BASEUPD macro that would update the base register with the
hidden offset value maintained by the SIGUPD macro (this is needed for
tests that use SIGUPD inside loops),
Here, though,  I can see that an OFFSETUPD macro that manually sets the
hidden offset instead would work better.
All of these macros should check that base+offset is within bounds.

PR #101 (2020-03-25): Restructure
Changes in this pull-request:

    * restructuring the riscv-test-suite to indicate clearly what is deprecated, wip and usable
      tests.
    * based on the above fixed the directory structure for riscv-targets where-ever applicable. Only
      tested riscvOVPsim and spike.
    * fixed script bugs for spike as well
    * renamed rv32i/I-IO.S to rv32i/I-IO-01.S along with necessary changes to the reference files
      and Makefrag
    * renamed mbadaddr csr to mtval as raised in issue #31
    * C.SWSP-01.S test updated to fix issue #37

Close: #8 
Close: #31 
Close: #30 
Close: #3 [... cut]
closed 2020-03-25 by neelgala