riscv/riscv-arch-test #38: I-LB-01 test - Load the data into X0 GPR register

santhoshvlsi, 2019-01-31:
From the RISC-V specification, Load the data into **X0** register, it can be illegal instruction. you can change the code into below format
    # Addresses for test data and results
    la      x21, test_C_data
    la      x22, test_C_res
    # Test
    lbu     **x1**, 0(x21)
    # Store results
    sw      **x1**, 0(x22)
    RVTEST_IO_ASSERT_GPR_EQ**(x1**, 0x00000000)

aswaterman, 2019-01-31:
The RISC-V spec does _not_ say that LBU with a destination of x0 can raise the illegal-instruction exception.

eroom1966, 2020-02-26:
This is perfectly legal behavior - not  a bug

closed 2020-02-26 by eroom1966

talhashahzad12345, 2024-04-19:
RiscV User Level ISA V2.2 says: 
Loads with a destination of x0 must still raise any exceptions and action any other side effects even though the load value is discarded.

Can someone please explain why would this be legal then?

aswaterman, 2024-04-19:
Two reasons: regularity (other seemingly useless instructions that write x0 are also legal), and because it's occasionally useful (e.g. if you have an MMIO region that has side effects on loads, you can use a load to x0 if you want to action the side effects but don't care about the data value).