riscv/riscv-arch-test #45: Reorganization of test suites for code maintainability

debs-sifive, 2019-02-13:
I am curious if there is any intention to reorganize the test suites. Right now, this seems a bit superficial, but in the long run this may have an effect on this repo's maintainability. For example, the [`rv32si`](https://github.com/riscv/riscv-compliance/tree/master/riscv-test-suite/rv32si) and the [`rv32mi`](https://github.com/riscv/riscv-compliance/tree/master/riscv-test-suite/rv32mi) suites both have separate subdirectories called `rv64si` which appear not to be symlinked. This has so far caused at least one issue (#44). Given this, 

- why are symlinks not being used?
- why do these `rv64*` subdirectories not exist as their own suites? I.e., why is the `rv64si` directory inside of `rv32si`, for example, not its own test suite?
- why does this layer of indirection exist in the `rv32*` directories? Is the intention to eventually pull the `rv64*` tests out into their own suites?

Note: Beyond this superficial reorganization, some discussion of privileged/unprivileged test organization has occurred in #40.

jstraus59, 2019-02-13:
Today the rv32i, rv32im, rv32imc, rv64i, rv64im are in the right shape

the others were quick ports of the riscv-isa-tests and hence too many
macros, shared use between rv64 and rv32 and many includes - and they dont
use any of the logging and assertion macros and do quite a simple signature
based don pass fail in the test - so they in fact all need restructuring to
adhere to the current as documented - they are all old style and were
included as a simple quick and easy way of seeing who good the old,
to-be-deprecated riscv-isa-tests

I think maybe we could move these badly styled ones into another directory
so they can be used. but so they don't confuse. eg:

riscv/riscv-compliance/riscv-test-suite [rv32i, rv32im, rv32imc, rv64i,
rv64im]
riscv/riscv-compliance/riscv-test-suite-work-in-progress   [rv32mi, rv32si,
rv32ua, rv32uc, rv32ud, rv32uf, rv32ui]

thanks
Simon

-- 
====================================================================
The information contained in this electronic mail message and any
attachments hereto
is privileged and confidential information intended only for the use of the
individual or
entity named above or their designee.  If the reader of this message is not
the intended
recipient, you are hereby notified that any dissemination, distribution or
copying of this
communication is strictly prohibited.  If you have received this
communication in error
please immediately notify us by return  message or by telephone and delete
the
original message from your mail system.  Thank you.
====================================================================

allenjbaum, 2019-05-28:
The directory structure is pretty screwed up in my opinion.
The directories mentioned by Simon are mostly OK, though it isn't obvious that rv64i has only the unprivileged ops that exist in RV64i but don't exist in RV32i (i.e. all of the W ops). 
On the other hand, all of the other rv32 tests have rv64 subdirectories inside of them, and those subdirectories should exist at the top level, not be a subdirectory. The feramework can make sure both are executed, but it would be a bit more clear if rv64i included rv32i tests (with the appropriate variable set, just as the rv32ux tests include rv64x tests (but at the wrong directory level)

allenjbaum, 2019-12-14:
This has been completely revamped under the new testformat spec  and the riscof framework matches the new directory structure (or vice versa).
I'm not sure we want to close this now or when we get to v.2 of the spec

allenjbaum, 2020-12-16:
The recent merge of tests have completely revamped the directory structure as described in the test spec, and in riscv-compliance/riscv-test-suite/README.md:

allenjbaum, 2021-04-27:
The tests suite directories have been reorganized, so I'm closing this issue. Please reopen if you think this has not been done correctly. The current directory structure should match the test spec format now.

closed 2021-04-27 by allenjbaum