riscv/riscv-arch-test #47: Machine  mode atomic extension tests?

aprnath, 2019-02-17:
The Atomic suite in this repository is labeled 'ua'. Are there any machine mode only Atomic extension compliance suite tests?

allenjbaum, 2019-02-19:
Other will correct me, for sure, but:
The only difference in a machine mode vs. user mode test for this is
whether this is executed in VM mode with address translation (with the
implication it could trap).
Since this test is in the unprivileged test suite, it shouldn't require
privileged resources (and, indeed, the tests themselves don't),  and
shouldn't cause traps (which require privileged mode resources).
So this may be in user mode test, but it should (or could) be operating in
a bare physical-only mode or under VM and either in Mmode or uMode.
Having said that: I would expect similar tests in the privileged test suite
that would add test that get page faults, etc.

neelgala, 2020-03-19:
The ua tests will be deprecated for now. They were directly borrowed from the riscv-tests repo for reference purposes. Dedicated tests for the A extension (machine, supervisor, user) are yet to be developed by the compliance group.

PR #101 (2020-03-25): Restructure
Changes in this pull-request:

    * restructuring the riscv-test-suite to indicate clearly what is deprecated, wip and usable
      tests.
    * based on the above fixed the directory structure for riscv-targets where-ever applicable. Only
      tested riscvOVPsim and spike.
    * fixed script bugs for spike as well
    * renamed rv32i/I-IO.S to rv32i/I-IO-01.S along with necessary changes to the reference files
      and Makefrag
    * renamed mbadaddr csr to mtval as raised in issue #31
    * C.SWSP-01.S test updated to fix issue #37

Close: #8 
Close: #31 
Close: #30 
Close: #3 [... cut]
closed 2020-03-25 by neelgala