riscv/riscv-arch-test #52: Why adding LOCAL_IO_PUTC(a0) in tests? What is the checking?

MRDore, 2019-03-14:
3 months back I downloaded and have been running compliance tests, things were fine. 
Updated test have LOCAL_IO_PUTC(a0); in the test case causing failure. 
What is the expectation here from the test case?  What is testing here by this in all test cases? Please comment on this.

```
FN_WriteStr:
    mv          t0, a0;
10000:
    lbu         a0, (t0);
    addi        t0, t0, 1;
    beq         a0, zero, 10000f;
    **LOCAL_IO_PUTC(a0);**
    j           10000b;
10000:
    ret;
```

eroom1966, 2019-03-14:
Can you also describe the failure you are seeing
- Compilation ?
- Execution ?
- Signature Comparison ?

Can I ask which target you have been using to execute the tests ?
- riscvOVPsim ?
- rocket ?
- spike ?
...

There are 5 macros which need defining for each target, and it is acceptable for these macros to have no code associated at all. so in the following file 
riscv-target/riscvOVPsim/compliance_io.h
you will see that defining 'RVTEST_IO_QUIET' creates macros which are NULL for the following

#define RVTEST_IO_INIT
#define RVTEST_IO_WRITE_STR(_SP, _STR)
#define RVTEST_IO_CHECK()
#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)
#define RVTEST_IO_ASSERT_SFPR_EQ(_F, _R, _I)
#define RVTEST_IO_ASSERT_DFPR_EQ(_D, _R, _I)

otherwise they associate a number of IO and ASSERT capabilities

Thx
Lee

MRDore, 2019-03-15:
Hi Lee

1. When I run with the target specified to spike, I am getting error as follow and not generating tests otherthatn rv32i. Could you tell me anything should I need to taken care here (I am using command make -d RISCV_TARGET=spike & make RISCV_TARGET=spike -isa=rv32imc).
_FAIL: 1/55
Reaping losing child 0x00769030 PID 80377 
make[1]: *** [verify] Error 1
Removing child 0x00769030 PID 80377 from chain.
make[1]: Leaving directory `/wdc/proj/riscv/mustang/users/dore.m.r/compliance_test/26feb/riscv-compliance/riscv-compliance'
Reaping losing child 0x0178eb30 PID 79430 
make: *** [all_variant] Error 2
Removing child 0x0178eb30 PID 79430 from chain._

2. When I generate with default target like **make all_variant** and when I ran with  I see the execution error(self-check fail) as follow

UVM_INFO [tracer] @406400.0ns: => @406400 #80 C0 8000003c [5391e193] ori      x3, x3, 0x539         OK
UVM_INFO [tracer] @415900.0ns: =>     x30 = 8000a040
UVM_INFO [tracer] @415900.0ns: => @415900 #81 C0 80000040 [0000af17] auipc    x30, 0xa OK
UVM_INFO [tracer] @416000.0ns: =>     STADDR = 8000a000       STDATA = 00000539
UVM_INFO [tracer] @416000.0ns: => @416000 #82 C0 80000044 [fc3f2023] sw       x3, -0x40(x30)      OK
UVM_INFO [tracer] @416000.0ns: => Detected store tohost with value = 00000539 .. stopping checks (83 instructions, 82 checked)
UVM_ERROR (0) @ 416000.0ns: uvm_test_top.riscv_env.tracer [tracer] Self-Check FAIL: tohost exit code != 1 (539)

Actually, As per your comment, I need things specific only to spike(our target is a spike) that can remove the macro that is not needed for us. 

Thanks & Regards
Dore M R

eroom1966, 2019-03-15:
Hi Dore M R

can you try executing 
$ make RISCV_TARGET=spike

Please provide the full log including STDOUT and STDERR
The errors you are showing look like system reported errors, I do not even see the applications being compiled.

What machine are you running ?
Finally, if you just type 
$ spike

what is printed ?

Thx
Lee

MRDore, 2019-03-15:
1. I tried executing both 
   $ make clean
   $ make RISCV_TARGET=spike. same fail.
   added the run log in attachment, please find it.

2. our spike and compilers are set. didn't get how to define compiler name.
spike path: /wdc/apps/riscv/bin/spike
3. $spike 
 __usage: spike [host options] <target program> [target options]
 Host Options:
   -p<n>                 Simulate <n> processors [default 1]
   -m<n>                 Provide <n> MiB of target memory [default 2048]
   -m<a:m,b:n,...>       Provide memory regions of size m and n bytes
                           at base addresses a and b (with 4 KiB alignment)
   -d                    Interactive debug mode
   -g                    Track histogram of PCs
   -l                    Generate a log of execution
   -h                    Print this help message
   -H                 Start halted, allowing a debugger to connect
   -C                 Enable log & commitlog
   --isa=<name>          RISC-V ISA string [default RV64IMAFDC]
   --pc=<address>        Override ELF entry point
   --hartids=<a,b,...>   Explicitly specify hartids, default is 0,1,...
   --ic=<S>:<W>:<B>      Instantiate a cache model with S sets,
   --dc=<S>:<W>:<B>        W ways, and B-byte blocks (with S and
   --l2=<S>:<W>:<B>        B both powers of 2).
   --extension=<name>    Specify RoCC Extension
   --extlib=<name>       Shared library to load
   --rbb-port=<port>     Listen on <port> for remote bitbang connection
   --dump-dts  Print device tree string and exit__
[fail.log.txt](https://github.com/riscv/riscv-compliance/files/2970833/fail.log.txt)

eroom1966, 2019-03-15:
By the look of your log, the test for I-MISALIGN_LDST-01 is failing (all others are passing)
what version of spike are you running, maybe spike has been changed in the way it handled misaligned ld/st ?

out of interest what happens if you run with the OVPsim simulator ?
$ make RISCV_TARGET=riscvOVPsim

eroom1966, 2019-03-15:
@MRDore 
Going back to your initial issue, you mention that you think that 
LOCAL_IO_PUTC(a0)
is causing a problem

Are you still thinking this is the case ?
because it looks like only one test is failing as I described above.

thx
Lee

MRDore, 2019-03-15:
I don't know the exact version, we installed on 28 -Jan-2018. 

@eroom1966 
FN_WriteStr:
    mv          t0, a0;
10000:
    lbu         a0, (t0);      **// I think  problem is here**
    addi        t0, t0, 1;
    beq         a0, zero, 10000f;  // **expecting 0 as a output from the memory but its not**
    LOCAL_IO_PUTC(a0);
    j           10000b;
10000:
    ret;

eroom1966, 2019-03-15:
@MRDore 
But this does not make sense, this code is never executed if the target is spike ?
the file that defines these macros are under 
riscv-target/soike/compliance_io.h

all the macros are defined as empty, meaning they compile no code.
```

#ifndef _COMPLIANCE_IO_H
#define _COMPLIANCE_IO_H

//-----------------------------------------------------------------------
// RV IO Macros (Non functional)
//-----------------------------------------------------------------------

#define RVTEST_IO_INIT
#define RVTEST_IO_WRITE_STR(_SP, _STR)
#define RVTEST_IO_CHECK()
#define RVTEST_IO_ASSERT_GPR_EQ(_SP, _R, _I)
#define RVTEST_IO_ASSERT_SFPR_EQ(_F, _R, _I)
#define RVTEST_IO_ASSERT_DFPR_EQ(_D, _R, _I)

#endif // _COMPLIANCE_IO_H
```

the code fragments you have referenced are for the riscvOVPsim target, not the spike target.
thx
Lee

MRDore, 2019-03-15:
**@eroom1966** 
Maybe in November, downloaded the compliance and I used the command to **make all_variant**, test cases generated without those macros and worked fine with our simulator & spike. 
Now I wanted to test the updated compliance tests with the same and followed the same process got stuck.
Could you tell me how to generate updated compliance test cases with the target as a spike?

eroom1966, 2019-03-15:
@MRDore 

I think I am a little confused, so would like to clarify some information you have supplied

If you use the command 
$ make all_variant

This will compile code (by default) specifically for riscvOVPsim, including the use of the macros from here
riscv-target/riscvOVPsim/compliance-io.h

If you wish to compile for spike, then the command should be 
$ make RISCV_TARGET=spike all_variant 
This will compile code using the macros from here
riscv-target/spike/compliance-io.h
using this flow, the code fragment you referred to in your original post for **FN_WriteStr**, should not exist, if it does there is an error in the compilation

 if you are targeting your own simulator

> those macros and worked fine with **our simulator** & spike.
you need to define those macros, probably the same as spike, and again, if there is a call to **FN_WriteStr** in the resulting elf file, this is probably an error.

Finally, from the log you provided earlier, only one test failed on spike, which is I-MISALIGN_LDST-01, do you agree ?

If so this may be because of the SPEC changes for the 2.3 ratification has changed the way spike handles the misaligned load and stores - this will need investigating

Thx
Lee

MRDore, 2019-03-15:
**@eroom1966** 
I agree only one test case failed in rv32i, still more tests are there to generate right, why it stops the test generation? or is there any switch to control this?

Instead of the spike, I wish to target our simulator, is it possible with the compliance environment?

Thanks 
Dore M R

eroom1966, 2019-03-15:
@MRDore 

> I agree only one test case failed in rv32i, still more tests are there to generate right, why it stops the test generation? or is there any switch to control this?

The test generation/execution runs in groups, so you can manually run the other groups individually
$ make RISCV_TARGET=spike RISCV_ISA=<group>

It would be worthwhile for the group if you could debug why spike has changed behavior, and whether this is correct.

> Instead of the spike, I wish to target our simulator, is it possible with the compliance environment?

Yes, simply add your simulator as a new target, and define the macros in 
riscv-target/<your-sim>/compliance_io.h

These macros can be defined as empty, they are only required for additional debug capabilities

Thx
Lee

MRDore, 2019-03-15:
@eroom1966 
$make clean
$ make RISCV_TARGET=spike RISCV_DEVICE=rv32ua, this is not working I tried all groups, it took an integer(rv32i group) first and generation fails with the same error as follow.  is this due to our spike issue? Could you provide me with the passed logs which targeted spike and tested with individual groups other than rv32i(don't want integer)? 
_

> > ffffffff
> > ffffffff
>  ... FAIL
> Check         I-NOP-01 ... OK
> Check          I-OR-01 ... OK
> Check         I-ORI-01 ... OK
> Check     I-RF_size-01 ... OK
> Check    I-RF_width-01 ... OK
> Check       I-RF_x0-01 ... OK
> Check          I-SB-01 ... OK
> Check          I-SH-01 ... OK
> Check         I-SLL-01 ... OK
> Check        I-SLLI-01 ... OK
> Check         I-SLT-01 ... OK
> Check        I-SLTI-01 ... OK
> Check       I-SLTIU-01 ... OK
> Check        I-SLTU-01 ... OK
> Check         I-SRA-01 ... OK
> Check        I-SRAI-01 ... OK
> Check         I-SRL-01 ... OK
> Check        I-SRLI-01 ... OK
> Check         I-SUB-01 ... OK
> Check          I-SW-01 ... OK
> Check         I-XOR-01 ... OK
> Check        I-XORI-01 ... OK
> --------------------------------
> FAIL: 1/55
> make[1]: *** [verify] Error 1
> make[1]: Leaving directory `/wdc/proj/riscv/mustang/users/dore.m.r/compliance_test/26feb/riscv-compliance/riscv-compliance'
> make: *** [all_variant] Error 2

_

Spike issue I will check and I will try by adding our simulator as a target. 

Thanks & Regards
Dore M R

eroom1966, 2019-03-18:
@MRDore 

make RISCV_TARGET=spike **RISCV_DEVICE=rv32ua** RISCV_ISA=rv32ua

MRDore, 2019-03-18:
Thanks, @eroom1966.

closed 2019-03-18 by MRDore