riscv/riscv-arch-test #58: I-FENCE.I-01 Spike execution

amerlo94, 2019-05-20:
In running the regression on Spike, I've come across a mismatch in the Spike execution. In particular for the I-FENCE.I-01 test at line 67 (reference below), Spike executes:

(0x001101b3) instead of (0x00000137)

It is something related on how Spike implements the FETCH instruction?

https://github.com/riscv/riscv-compliance/blob/2657d67b049af1dc4bff40e82d652ed0f2ef9aee/riscv-test-suite/rv32i/src/I-FENCE.I-01.S#L67

allenjbaum, 2019-05-20:
I don't think I understand your issue.
The LUI instruction (encoding 0x00000137 ) should be overwritten with the
instruction at line 63 and stored at line 64, fenced at line 65 which makes
sure that the store took effect before the fetch of the instruction at 67.
So, when line 67 is executed, it should execute
add x3, x2, x1
and not the original
lui x2, 0
assuming the fence.i is working correctly

amerlo94, 2019-05-20:
Hi @allenjbaum,

I think I got the instruction flow, thank for the quick reply.

closed 2019-05-20 by amerlo94