riscv/riscv-arch-test #63: Global linker script is not appropriate

jeremybennett, 2019-08-13:
We have a handful of linker scripts, in `riscv-test-env/{p,pm,pt,v}/link.ld`. These impose certain views on section naming and memory maps. These really are target platform specific details, and so should be configured on a per platform basis. Otherwise we'll be adding a new version for every platform to be tested.

eroom1966, 2019-11-05:
@jeremybennett 
What is your proposed fix?
for a physical platform - there is a real physical constraint, and in fact for ibex, you will see I did create a platform specific linker script
For simulators however, these can be any shape we wish, so there is no real restriction, just a common description for the linker memory layout and the simulation setup

jeremybennett, 2019-11-10:
I was really just suggesting they should be in the target specific directory, not centrally.

allenjbaum, 2020-02-25:
Can those environments be easily overridden? 
Where is the platform specific ibex script? 
Does that solution work for everyone (without breaking anything)?

neelgala, 2020-03-19:
This is requirement has been fixed in RISCOF. Each target provides their own linker scripts to the framework. A sample linker script is also provided by the initial setup command.

allenjbaum, 2021-04-27:
This has been fixed pre-riscof, I believe.

neelgala, 2021-04-28:
yes. can be closed. Open if issue still persists.

closed 2021-04-28 by neelgala