riscv/riscv-arch-test #67: RV32I Immediate Operands error

rongcuid, 2019-09-25:
I am making a new target of my RISC-V platform, and I encounter these compiler error in the test suite:

```
src/I-RF_x0-01.S:57: Error: illegal operands `xori x0,x0,0xFFFFF803'
src/I-JALR-01.S:291: Error: illegal operands `jalr x0,x3,0xFFFFFFFF'
```
And so on. I used Zephyr SDK 0.10.1, which provides GCC 8.3.0.

I believe that GCC is unhappy with the "sign extension" done by hand, as I could remove these errors by using corresponding negative numbers.

eroom1966, 2019-11-05:
OK, I have changed these tests, good news still works with my version compiler - I will issue a pull request

rongcuid, 2019-11-05:
Also, RV32I tests seems to fail on RV64I architectures due to OP and OP-32, OP-IMM and OP-IMM32 difference. Is that intentional?

allenjbaum, 2019-11-08:
Definitely not intentional. Could you be more specific on the errors you
are seeing so we can track them down?

rongcuid, 2019-11-08:
I might be wrong. Previously I tried to manually change all the invalid operands to negative numbers. I failed a few of them, and some are my fault. However, I think there are some problems in the tests too.

I failed: ENDIANESS, RF_SIZE, RF_WIDTH, MISALIGNED_JUMP, JAL-01, JALR-01, LUI-01, AUIPC, all loads, SB, SH, and some more.

Of these, LUI-01 is almost definitely not my fault:

```
# Test Begin

# Test part A - general test of LUI

= Assertion Violation =
== File ==
src/I-LUI-01.S
== Line ==
72
== Register ==
x5
== Got ==
0xfffffffffffff000
== Expected ==
0xFFFFF000
```
So is AUIPC:

```
# Test Begin

# Test part A - general test of AUIPC

= Assertion Violation =
== File ==
src/I-AUIPC-01.S
== Line ==
90
== Register ==
x6
== Got ==
0x0000000080000000
== Expected ==
0x00000000
```

LW:

```
# Test Begin

# Test part A1 - test base address + 0

# Test part A1  - Complete

# Test part A2 - test base address - 1

= Assertion Violation =
== File ==
src/I-LW-01.S
== Line ==
80
== Register ==
x25
== Got ==
0xfffffffff33344f4
== Expected ==
0xF33344F4
```
And more.

I notice that all the targets shipped pass a parameter to run the 32-bit variant when running the RV32I tests, and in my understanding, RV64I and RV32I are not really binary compatible (due to OP-32 and OP-IMM32). This leaves my 64-bit only implementation without the majority of tests.

Perhaps this should be a separate issue?

allenjbaum, 2019-11-08:
I'm guessing the assertions are set up for 32b operands and not 64b.
It is odd that this hasn't been reported previously. The signatures
probably match, which is the most important part, so this may be just a
false negative.
This will likely be fixed in future test suites as there will be completely
different tests for RV64 and RV32, and therefore different assertions (at
worst).

rongcuid, 2019-11-08:
I do not think the signatures match, especially for the few failures I listed above, RV64I probably needs its own set of tests that covers all instructions, separate from RV32I.

neelgala, 2020-03-18:
@rongcuid yes the rv64i tests do require their own set and are not complete. Only the rv32i, rv32im, rv32imc, rv32ifencei and rv32icsr are the ones that are currently supported. The 64 bit tests are under development.

closed 2020-03-25 by neelgala

MarcKarasek, 2020-10-15:
I do not see where the original problem with the xori 

src/I-RF_x0-01.S:57: Error: illegal operands `xori x0,x0,0xFFFFF803'
src/I-JALR-01.S:291: Error: illegal operands `jalr x0,x3,0xFFFFFFFF'

has been fixed?  I am seeing the same problem using a 64bit compiler. 

Also: What is the current state of the RV64 tests?

allenjbaum, 2020-10-15:
I dimly recall having seen this before, and that this is a toolchain issue.
If instead 0xFFFFF803 was encoded as -0x7FD, it would probably work.
Which toolchain are you using?

allenjbaum, 2020-10-15:
RV64 tests are coming, and we hope to have them merged by the end of next
month.

PR #101 (2020-03-25): Restructure
Changes in this pull-request:

    * restructuring the riscv-test-suite to indicate clearly what is deprecated, wip and usable
      tests.
    * based on the above fixed the directory structure for riscv-targets where-ever applicable. Only
      tested riscvOVPsim and spike.
    * fixed script bugs for spike as well
    * renamed rv32i/I-IO.S to rv32i/I-IO-01.S along with necessary changes to the reference files
      and Makefrag
    * renamed mbadaddr csr to mtval as raised in issue #31
    * C.SWSP-01.S test updated to fix issue #37

Close: #8 
Close: #31 
Close: #30 
Close: #3 [... cut]