riscv/riscv-arch-test #70: Target error exit condition

towoe, 2019-10-08:
I noticed that the tests do not fail immediately if there is an error in `$(TARGET_SIM)`.
So for example in [Ibex](https://github.com/riscv/riscv-compliance/blob/master/riscv-target/ibex/device/rv32imc/Makefile.include#L13) the termination is done by `;` and not `&&`.
I was just wondering if this a feature so all the other tests are also executed.
If that is the case maybe this can be improved by using `&&` and ignore the error with `-` in the calling Makefile so it is more obvious in the `make` output.
Or maybe this is not the intended way. I am happy to create a PR but wanted to check before what the intentions here are.

allenjbaum, 2019-12-14:
I believe the intention is (or should be) to run (or attempt to run) all tests, reporting failures for each. I have seen riscof have this kind of behavior, and it makes it easier to see a class of problems that might be fixed all at once instead of one by one.
Please close this if this answers your issue,

towoe, 2020-02-13:
Following the discussion in #80 (v2 of framework is riscof, change for all cores to be avoided) closing this issue.

closed 2020-02-13 by towoe