riscv/riscv-arch-test #74: Strange assembly instruction generated in I-LB-01.S

myw2s-git, 2019-11-06:
Hi

Our team is considering to use Syntacore SCR-1. Syntacore provided a pre-built toolchain and the version is 7.1.0. However, we compiled a newer version 8.1.0. Recently, we ran the RISC-V compliance test but we got 8 test cases failed which are:

compliance_I-LB-01
compliance_I-LBU-01
compliance_I-LH-01
compliance_I-LHU-01
compliance_I-LW-01
compliance_I-SB-01
compliance_I-SH-01
compliance_I-SW-01

After tracing the cause, we located the toolchain generated some strange assembly codes for the following pseudo-instruction:

la x10, test_A4_data + 2048

**8.1.0**
55c:       00000f97            auipc     t6,0x0   
560:       5a4f8f93             addi       t6,t6,1444 # b00 <begin_regstate>   
564:       000fa183            lw           gp,0(t6)   
568:       004fa203            lw           tp,4(t6)   
56c:       008fa283            lw           t0,8(t6)   
570:       b8418513           addi       a0,gp,-1148 # 12cc <_end+0x36c>

The correct address of x10 (a0) should be 0x12cc, however, the above codes make x10 to 0xFFFFFB84. This value causes ASSERT to the core and makes the test failed. So, I did 2 trial tests:

1. Use Syntacore pre-built toolchain (7.1.0) to compile the code
2. Modify the pseudo-instruction and compile with 8.1.0

The results are:

1. The assembly codes look prefect and all the tests are passed. As the compiler is different, the correct value of x10 (a0) is 0x12ec.

**7.1.0**

 570:	00001517          	auipc	a0,0x1
 574:	d7c50513          	addi	a0,a0,-644 # 12ec <_end+0x38c>

2. Modify from "la x10, test_A4_data + 2048" to the following and rebuilt with 8.1.0: 

la x10, test_A4_data
addi x10, x10, 1024
addi x10, x10, 1024

570:       00000517           auipc     a0,0x0   
574:       55c50513           addi       a0,a0,1372 # acc <test_A4_data>   
578:       40050513           addi       a0,a0,1024   
57c:       40050513           addi       a0,a0,1024

The value of x10 (a0) is 0x12cc which is correct. And, all the test cases are passed as well.

=============================================================
a. May I ask if the newer version of RISC-V GNC toolchain support this syntax "la rd, symbol + offset"?? 
b. Why the old version of the toolchain can generate correct assemble codes but not in newer version? Can you help to verify this issue for me?

Thank you very much!
Jason

aswaterman, 2019-11-06:
Newer versions of binutils support linker relaxation against the `gp` register to reduce code size.  It seems the compliance tests need to be issuing a sequence to initialize `gp`, such as:

```
.option push
.option norelax
        la gp, __global_pointer$
.option pop
```

An alternative solution is to use a linker script that avoids defining the `__global_pointer$` symbol; then, the linker will not attempt to perform the `gp` relaxation.

myw2s-git, 2019-11-06:
Hi

Thank you for your reply! It doesn't work if I just add the code snippet to the _start, because the gp will be modified in different places. I have to add right before "la x10, test_A4_data + 2048".

Therefore, I just simply add '-mno-relax' to the linker flags to disable it. The assembly codes are same as those generated by the old version compiler (7.1.0).   

On the other hands, if writing c program (not pseudo-instruction), is it just adding the gp initialization in _start? Does the compiler take care of the gp?

Jason

aswaterman, 2019-11-06:
In the normal use case, the C library's implementation of _start is responsible for initializing gp.

closed 2019-11-06 by myw2s-git