riscv/riscv-arch-test #76: I-AUIPC-01 test writes signature to wrong address

bobbl, 2020-01-19:
The following macro from `riscv-test-env/riscv_test_macros.h` 

    #define TEST_AUIPC(inst, destreg, correctval, imm, swreg, offset, testreg) \
        TEST_CASE(testreg, destreg, correctval, swreg, offset, \
          1: \
          inst destreg, imm; \
          la swreg, 1b; \
          sub destreg, destreg, swreg; \
          )

clobbers `swreg` to get an position independent result. But the macro `TEST_CASE` assumes that `swreg` points to the signature and therefore the result is written to somewhere in the code section, not the signature. Consequently, the reference signature of test `rv32i/I-AUIPC-01` is unchanged from the initial all 1s and always correct, no matter if `AUIPC` is correctly implemented or not.

I propose to add an additional argument to `TEST_AUIPC` for a register that can be clobbered. I will submit a pull request soon.

allenjbaum, 2020-01-20:
Do you need to add an extra argument? IF I'm reading this correctly,
TEST_AUIPC has the argument testreg - which only gets used as a temp
register in the assert macro, If I;m understanding this correctly. Just
substitute that one.

bobbl, 2020-01-20:
Nice idea. Much simpler than my solution. I tested it and modified the push request accordingly.

yupferris, 2020-02-06:
I've run into this as well, and the associated PR (https://github.com/riscv/riscv-compliance/pull/77) looks good to me. Any chance we could get this looked at/merged?

closed 2020-02-15 by bobbl