riscv/riscv-arch-test #84: I-SW-01 corrupts .text region

byllgrim, 2020-02-08:
Hi, sorry if this is actually a mistake on my end.
But please allow me to explain what i think is a bug in the [I-SW-01 test](https://github.com/riscv/riscv-compliance/blob/master/riscv-test-suite/rv32i/src/I-SW-01.S).

Brief:
The test writes to an address below the .data section, and program code is overwritten.

--------

More details:

Here is the first label that the test sets up within the .data section
```
# Input data section.
	.data

# Output data section.
RV_COMPLIANCE_DATA_BEGIN

test_1_res:
	.fill 5, 4, -1
```

In the beginning of the test, that address is saved to x5
```
    # Addresses for test results
	la	x5, test_1_res
```

Shortly afterwards, it is used in a store operation
```
    # Clear memory
    sw      x0, -0x4(x5)
```

This `-0x4` gambles about not storing into the .text section.
At least for me, it overwrote my halt-simulation code, so the test never ended.

--------

Why it might not be a problem:

One can set up linker scripts to place the data section at a safe distance.
Or one can set up the macros RV_COMPLIANCE_CODE_END and RV_COMPLIANCE_DATA_BEGIN to have some safety buffer.

yupferris, 2020-02-08:
This is certainly an issue; I also encountered this.

byllgrim, 2020-02-10:
I guess some `test_res_buffer` would make the problem go away:

```
# Output data section.
RV_COMPLIANCE_DATA_BEGIN

test_res_buffer:
	.word 0

test_1_res:
	.fill 5, 4, -1
```

(that's how I temporarily defined my RV_COMPLIANCE_DATA_BEGIN to get around this)

allenjbaum, 2020-02-24:
If this issue (#84) has been fixed-- can someone close it?

byllgrim, 2020-02-25:
> If this issue (#84) has been fixed-- can someone close it?

I just removed my local fix and re-ran tests.
Problems seem to be gone on my end.

(weird that github defaults to _not_ notify me about the fixes above)

byllgrim, 2020-02-25:
Although... @eroom1966 @allenjbaum 
Sorry, I was a bit to quick to reply.

The problem still persists in `I-SH-01` it seems.

----

```
        RVTEST_IO_WRITE_STR(x31, "# Test number 1\n")

    # Addresses for test results
        la      x5, test_1_res

    # Clear memory
    sh      x0, 0x0(x5)
    # Test
    li      x16, -0x1
    addi    x31, x5, 0
    sh      x16, 0x0(x31)

    #RVTEST_IO_ASSERT_GPR_EQ(x6, x31, test_1_res)
    RVTEST_IO_ASSERT_GPR_EQ(x6, x16, -0x1)
    # Clear memory
    sh      x0, -0x2(x5)
    # Test
    li      x15, 0x1
    addi    x30, x5, 0
    sh      x15, -0x2(x30)
```

At least for me, the test never halts, because that final line `sh x15, -0x2(x30)` overwrites my RV_COMPLIANCE_HALT code.
It is the very same problem that I had with I-SW-01.

allenjbaum, 2020-02-26:
could you reopen the issue #84? (with comments how it effects SH; probably
should look at the SB test as well)

byllgrim, 2020-02-28:
(sorry for my subpar usage of the tickets tracker 😛. Also I don't have time to thoroughly verify everything atm)

The problem that was fixed in I-SW-01, was a write relative to the very start of `.data` which ended up writing into `.text` hence overwriting my RV_COMPLIANCE_HALT.
It is now fixed for the SW test, but it is still a problem in SH (which I tested) and possibly also SB (which I presume).

In all cases, the first condition for the problem is around RV_COMPLIANCE_DATA_BEGIN (though this code is perfectly ok by itself):
```
RV_COMPLIANCE_CODE_END

# Input data section.
        .data

# Output data section.
RV_COMPLIANCE_DATA_BEGIN

test_1_res:
        .fill 5, 4, -1
```

The second part is when a negative offset is used when referencing `test_1_res`:
```
    la      x5, test_1_res
    [...]
    sh      x0, -0x2(x5)
```

byllgrim, 2020-02-28:
Disclaimer:
It _could_ be that some documentation forbids .text and .data to be so close that RV_COMPLIANCE_HALT is perfectly adjacent to RV_COMPLIANCE_DATA_BEGIN.
Then this would all be my own fault.
But I have not seen any such restrictions.

allenjbaum, 2020-02-28:
Just to be clear: you found a bug. The bug affects multiple tests, and has
since been fixed in one test, but *not* others(s).
You don't need to apologize, but probably the issue should be re-opened and
expanded to include the other tests, or another issue filed.

allenjbaum, 2020-03-17:
I don't suppose you could make a pull request with the fix(es) you want - that's the fastest way to get this resolved.

neelgala, 2020-03-18:
Having gone through the code I believe this is indeed a bug in the I-SH. 
The reason I did not see this is because spike uses a linker script where the <tohost> and <fromhost> labels occur between the text and data section. 

I just confirmed that the particular instruction sequence pointed out by byllgrim indeed does not have a footprint in the signature that spike generates. So definitely requires an update on the test.

@byllgrim is there a way I replicate this error at my end somehow?

byllgrim, 2020-04-06:
Hi.
Sorry I have not been up to date with github lately.

@allenjbaum sorry but I will prioritize other work so I won't be making a pull request any time soon.

@neelgala I'm not sure how you can _easily_ replicate it; I was using my own core. Maybe try to fiddle with the linker script of spike or some other implementation :)

neelgala, 2020-04-06:
@eroom1966 @simon5656  Could you guys look at re-generating/fixing this test?

allenjbaum, 2020-04-08:
I've been looking at the SH and SB tests. They are more broken than I
thought.
There are 7 test cases each with 5 tests, and each reserves 5 signature
words - but only uses offsets -4..+4 instead of 0..19
At the beginning of each test case, it loads the base (offset 0 of that
test), and uses it over and over (moving it from register to register for
coverage

So each test_case signature can bleed into the previous one, and
test_case1,  offset -4  is never tested since it's outside the signature
area.
So, I added a padding word in the signature area above the base of the
first test, but after DATA_BEGIN, and prefixed the reference_output with
the correct offset = -4 value that the first test_case produced.
I'm tempted to completely rewrite it by defining a macro (each test is
identical) , so each test would be a single line .
Each: starts with a base in (the same) register, then
 write 0  base+offset,
 loads an immediate,
  moves the sig_base to some other_base,
  write immediate to other_base+offset
  assert that the GPR with the immediate still has the immediate (why? to
make sure that the store didn't load?

I'll try to have a pull request ready before I go to bed tonight, and you
can test it (using git is still not my strong suit, so it will take a while)

allenjbaum, 2020-04-08:
This is the dead simple fix; I could also more extensively rewrite it the
way the SW test is written.
That one keeps incrementing the base, pre-adjusts it with some random
constant, stores using that base-random constant, then loads it back and
asserts the value.
Its much more comprehensive test, and if you write a custom test macro,
easier to auto-generate tests for coverage

allenjbaum, 2020-04-08:
pull#104 should fix this

allenjbaum, 2020-04-22:
Fixed with pull#104

neelgala, 2020-04-25:
pull #104 merged

closed 2020-04-25 by neelgala