riscv/riscv-arch-test #89: breakpoint test within rv32mi fails - missing register setting?

cr8601, 2020-02-14:
Hi,
I executed breakpoint test on sweRV core (https://github.com/chipsalliance/Cores-SweRV) and it failed. After some digging and asking the community (https://github.com/chipsalliance/Cores-SweRV/issues/33) I found out, that mstatus.mie must be set in order for success full execution. Therefore I simply added the following lines to the test:

```
# set mstatus.MIE=1 (enable M mode interrupt)
li      t0, 8
csrrs   zero, mstatus, t0
```
After that the test passes.

This behavior is described in Section 5.1, ‘Native M-Mode Triggers’ of the RISC-V External Debug Support specification. 
Do you agree and would you add this to the test?
Best regards
cr

allenjbaum, 2020-02-14:
> digging and asking the community (chipsalliance/Cores-SweRV#33

which test environment did you use?

cr8601, 2020-02-17:
Hi,
I am using "riscv-test-env/p". To be honest, I was not aware of test-env until now. It was just used as default after my intital target setup... 
But I also do not see anything specific to mstatus register within the other test-environments (v, pt, or pm).
regards
cr

allenjbaum, 2020-02-17:
Ah - so this is in a system that implements M-mode only, and also
implements the *optional* behavior of preventing breakpoint triggers (which
are exceptions, NOT interrupts) when mstatus.mie=0.
This is a rather obscure optional behavior (i.e. does not appear in the
priv mode spec, but only in the debug spec), so not too surprising that it
hasn't been caught before.
This specific behavior should not occur in cores that implement U-mode, and
is optional even in systems that don't.
So, this simple test change that enables sweRV to pass would fail for other
cores that expect it to trap even if mstatus.mie=0.
This is an example of optional behavior that requires either
 -  the test to be re-written to have different behavior depending on a
YAML parameter passed into the tests
 -  have two different tests which are selected by a YAML parameters.
Unlike other optional behavior (e.g. unaligned support) where tests can be
written that will correctly handle either case, there is no way to know
which optional behavior is correct without something like that YAML
parameter.
I'm not quite sure how to fix this for the v.1 of the compliance suite.
Anyone else have any ideas? Should this be fixed by having a separate sweRV
environment that sets mstatus.mie? (but then, the test won't check that the
trigger fails to trap when mstatus.mie is clear)

This should not be a problem for v.2, but the test will need to be
rewritten to test for both mstatus.mie behaviors, and the formal model will
need to be configured for the optional behavior.

cr8601, 2020-02-20:
Hi,
thank you for your detailed explanation. From my point of view I don't need a seperate sweRV env right now. Maybe I'll have another look at it once v.2 with configuration options is available.
Thanks and best regards
cr

closed 2020-02-20 by cr8601