riscv/riscv-arch-test #90: Report target execution error

towoe, 2020-02-14:
If a target simulator (defined with `TARGET_SIM`) fails to execute, this error should be reported clearly.
A reason for an error could be that the path to the simulator is incorrect, or an argument causes an error.
Currently each test is executed without reporting the failure of execution.
The signature comparison should fail, if no previous results are present in the output folder.
Not reporting the actual error can cause some misunderstanding to the actual problem.

@allenjbaum as discussed in #70 and #80

neelgala, 2020-03-18:
The issues you raise have been fixed in RISCOF (next version of the compliance framework). There, each target is expected to provide their own plugin which includes: the environment of execution, arguments, method of execution and thus the target-authors have complete control on the kind of errors they would like to report/halt on.

allenjbaum, 2020-12-17:
The new framework (re-RISCOF) has been merged in, along with new tests.
In it, environment files are decoupled to give control to the target on what it can do ( so environment specific Makefile.include is split from root Makefile ) You now have the ability to modify the target Makefile.includes to be more verbose and add more checks there. Please take a look at the Makefile and Makefile.include examples . If you have a specific change you would like to see in the non-target specific Makefile, please respond here.

allenjbaum, 2021-04-27:
I am assuming, based on the above comments, that this has been fixed for your purposes, so I'm closing this. If not, please reopen.

closed 2021-04-27 by allenjbaum