riscv/riscv-arch-test #96: how do I run all of the 64-bit tests, such as those in riscv-test-suite/rv32ui/rv64ui, through, say, spike ?

dsw, 2020-03-06:
I find that when I run the spike tests, not all of the 64-bit tests run.  All that seems to run are the 64-bit tests having a "W" in the instruction name, that is, those that are not the namesake of a 32-bit test, such as those in riscv-test-suite/rv64i/src.  In contrast, most of the 64-bit instructions that are the namesake of a 32-bit instruction do not seem to be tested.

I find the following directory that seems to contain 64-bit tests: riscv-test-suite/rv32ui/rv64ui .  This does not seem to participate in the testing of spike.  How come?  I cannot find how to run these tests.  How do I run all of the 64-bit tests through, say, spike?

Am I missing something obvious here?  I could just find a way to hack this in, but it seems that there must be an official way to do this and I would rather use that if there is one.

Perhaps this is related to https://github.com/riscv/riscv-compliance/issues/30 , which seems to suggest that the 64-bit tests are not even finished?

allenjbaum, 2020-03-11:
https://github.com/riscv/riscv-compliance/tree/master/riscv-test-suite/readme.md lists the standard test suites that are supported. The rv64 tests were quickly ported from some older tests, but are not supported. Most of these have been duplicated in the rv64i tests, so they should be redundant.

You could add them anyway by 
    adding a directory /riscv-target/spike/device/rv64ui and copying the /rv64i/Makefile.include into it , 
   copying a directory riscv-compliance/riscv-test-suite/rv64i  into /rv64ui, 
   modifying the /rv64ui/makefrag file  to include all the tests you want to run, 
   creating reference results in the /references subdirectory
and even then, it isn't clear that this will work.

pdonahue-ventana, 2020-03-11:
That link is wrong due to case-sensitivity.  Correct link:
https://github.com/riscv/riscv-compliance/blob/master/riscv-test-suite/README.md

allenjbaum, 2020-03-11:
I knew I should have checked that. Sloppy. Thanks.

neelgala, 2020-03-18:
the rv64i test suite is not complete and still a WIP. Suggest you look at riscv-tests by andrew for a good starting point for 64-bit tests.

PR #101 (2020-03-25): Restructure
Changes in this pull-request:

    * restructuring the riscv-test-suite to indicate clearly what is deprecated, wip and usable
      tests.
    * based on the above fixed the directory structure for riscv-targets where-ever applicable. Only
      tested riscvOVPsim and spike.
    * fixed script bugs for spike as well
    * renamed rv32i/I-IO.S to rv32i/I-IO-01.S along with necessary changes to the reference files
      and Makefrag
    * renamed mbadaddr csr to mtval as raised in issue #31
    * C.SWSP-01.S test updated to fix issue #37

Close: #8 
Close: #31 
Close: #30 
Close: #3 [... cut]
closed 2020-03-25 by neelgala