riscv/riscv-arch-test #97: in release 0.1, file riscv-test-suite/rv64i/Makefrag, variable rv64i_sc_tests contains SRAIW twice

dsw, 2020-03-06:
in release 0.1, file riscv-test-suite/rv64i/Makefrag, variable rv64i_sc_tests contains "SRAIW" twice; this is still the case in commit 13e1b57b52f1c0708bc6442a9668c5c01b4eb79e

> Makefile:49: target 'SRAIW.elf' given more than once in the same rule

dsw, 2020-03-11:
I think I fixed this and another similar bug in this pull request:
https://github.com/riscv/riscv-compliance/pull/98

however, see the note in that pull request that this change is tested in release 0.1 but not in HEAD as running the same tests in HEAD that I do in release 0.1 is not working

allenjbaum, 2020-03-11:
Thanks very much for making the pull request; every little bit helps.

neelgala, 2020-03-19:
merged and fixed. I have tested riscvOVPsim and spike and they seem to pass.

closed 2020-03-19 by neelgala