riscv/riscv-arch-test #99: I-MISALIGN-JMP-01

fsiegle, 2020-03-17:
In I-MISALIGN-JMP-01, mcause is written into the signature (e.g. second value of signature). The provided signature file assumes that mcause is zero (instruction address misaligned). While this makes sense, the misaligned address also triggers another synchronous exception 'illegal instruction' in my implementation because some nonsense instruction is read from that misaligned address.  According to RISC-V Privileged Architectures V20190608, Table 3.7 on page 38, the 'illegal instruction' exception has a higher priority than the 'instruction address misaligned' exception as is therefore reported, leading to a failing test.

neelgala, 2020-03-18:
Looks like you might be detecting the Instruction address misaligned exception when you perform the fetch. But the spec says that the Instruction address misaligned is generated by the branch/j/jalr instructions which generate the new address (typically in the execute stage). This is what the test also checks. 

So if you have decoded the j/jalr/branch instruction correctly (i.e. no illegal instruction), when you calculate the effective address and if it is mis-aligned then that instruction is tagged with the exception and the new PC should be the one that is present in the mtvec (which in the said test points to the label: _trap_handler). Note that the PC in the mtvec is valid and cannot be mis-aligned and thus the instruction fetched from that PC also cannot be illegal. Thus the situation you are describing is not possible with the current tests. 

hope this helps

fsiegle, 2020-03-19:
Good point, thank you very much! I should read the comments in the spec more carefully... As you pointed out, in my case it is the fetching mechanism that triggers the exception and not the branch logic.

closed 2020-03-19 by fsiegle