riscv/riscv-arch-test #102: Build failure src/I-MISALIGN_JMP-01.S:282: Error: Instruction csrr requires absolute expression

tommythorn, 2020-03-31:
Host: CentOS Linux 7

Prior to the most recent change:

> commit 5a978cfd444d5e640150d46703deda99057b2bbb
> Author: Neel Gala <neelgala@gmail.com>

a fresh clone would build and run with make without any configuration.  It seems the most recent change regressed something (I haven't triaged it yet):

```
$ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/tommy/install/20200211/libexec/gcc/riscv64-unknown-elf/7.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/tommy/boom-template/riscv-tools/riscv-gnu-toolchain/build/../riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/home/tommy/install/20200211 --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/tommy/install/20200211/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --src=../../riscv-gcc --enable-checking=yes --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medlow'
Thread model: single
gcc version 7.2.0 (GCC) 

$ make
for isa in rv32i rv32im rv32imc rv32Zicsr rv32Zifencei; do \
        make -j8 --max-load=4 RISCV_TARGET=riscvOVPsim RISCV_TARGET_FLAGS="" RISCV_DEVICE=$isa RISCV_ISA=$isa variant; \
                rc=$?; \
                if [ $rc -ne 0 ]; then \
                        exit $rc; \
                fi \
done
make[1]: Entering directory `/home/tommy/riscv-compliance'
make[1]: warning: -jN forced in submake: disabling jobserver mode.
make -j8 --max-load=4 \
        RISCV_TARGET=riscvOVPsim \
        RISCV_DEVICE=rv32i \
        RISCV_PREFIX=riscv64-unknown-elf- \
        run -C /home/tommy/riscv-compliance/riscv-test-suite/rv32i
make[2]: Entering directory `/home/tommy/riscv-compliance/riscv-test-suite/rv32i'
make[2]: warning: -jN forced in submake: disabling jobserver mode.
Compile /home/tommy/riscv-compliance/work/rv32i/I-ENDIANESS-01.elf
Compile /home/tommy/riscv-compliance/work/rv32i/I-RF_x0-01.elf
Compile /home/tommy/riscv-compliance/work/rv32i/I-RF_size-01.elf
Compile /home/tommy/riscv-compliance/work/rv32i/I-RF_width-01.elf
Compile /home/tommy/riscv-compliance/work/rv32i/I-MISALIGN_JMP-01.elf
Compile /home/tommy/riscv-compliance/work/rv32i/I-MISALIGN_LDST-01.elf
Compile /home/tommy/riscv-compliance/work/rv32i/I-DELAY_SLOTS-01.elf
Compile /home/tommy/riscv-compliance/work/rv32i/I-JAL-01.elf
src/I-MISALIGN_JMP-01.S: Assembler messages:
src/I-MISALIGN_JMP-01.S:282: Error: Instruction csrr requires absolute expression
src/I-MISALIGN_JMP-01.S:287: Error: Instruction csrr requires absolute expression
src/I-MISALIGN_LDST-01.S: Assembler messages:
src/I-MISALIGN_LDST-01.S:198: Error: Instruction csrr requires absolute expression
riscv64-unknown-elf-objdump: riscv64-unknown-elf-objdump: '/home/tommy/riscv-compliance/work/rv32i/I-MISALIGN_LDST-01.elf': No such file'/home/tommy/riscv-compliance/work/rv32i/I-MISALIGN_JMP-01.elf': No such file
```

neelgala, 2020-03-31:
Hi,
Thanks for the catch.
I had renamed the tests to following the general convention. 
I had forgotten to push the updated makefrag files. 

Will create a pull-request for the same.

neelgala, 2020-03-31:
my bad.. responded too soon.. the makefrag files are infact updated...

the only changes I have done is to rename the csr ``mbadaddr`` to ``mtval`` which is what is being followed since priv 1.10. 

The error you are seeing is because your toolchain isn't able to recognie mtval as a valid
csr name (it might be expecting mbadaddr). I believe an upgrade to a more recent version of the gnu-toolchain (>8.0) should fix the issue. Would that be something you can try out?

I don't see this error at my end - using riscv-gcc > 8.3.0.

tommythorn, 2020-03-31:
Ok that makes sense. Sorry about that. It looked like a simple regression and I didn’t realize my tools were an old. Thanks.

closed 2020-03-31 by tommythorn