riscv/riscv-arch-test #103: 'SB' and 'SH' tests expect to store data in addresses lower than `RV_COMPLIANCE_DATA_BEGIN`.

WRansohoff, 2020-04-07:
The RV32I 'store byte' and 'store halfword' instruction tests expect that they will be able to store data at a negative offset from the `test_1_res` label, which is located immediately after the `RV_COMPLIANCE_DATA_BEGIN` label:

SB: https://github.com/riscv/riscv-compliance/blob/master/riscv-test-suite/rv32i/src/I-SB-01.S#L66

SH: https://github.com/riscv/riscv-compliance/blob/master/riscv-test-suite/rv32i/src/I-SH-01.S#L66

I believe that this can (and should) cause failures if the `RV_COMPLIANCE_DATA_BEGIN` section is placed at the very beginning of RAM. If RAM starts at 0x20000000, these instructions attempt to store data in address 0x1FFFFFFD, which will likely fail or trigger a trap.

If I change the linker script to start the RAM address space at 0x20000004, the tests pass as expected. Should an extra word of data be added before the `test_1_res` section if a negative offset will be used with that label?

Thanks.

neelgala, 2020-04-07:
I think this is similar to the open issue #84 ?

WRansohoff, 2020-04-07:
Oh - yeah, that does look like it's caused by the same behavior. Sorry I didn't read more carefully.

closed 2020-04-07 by WRansohoff