riscv/riscv-arch-test #107: Clang/LLVM does not support all CSRs used in the compliance test suite

jeremybennett, 2020-04-22:
Clang/LLVM does not support as many CSRs as GCC. We can see this if we compile one of the tests explicitly:
```
riscv32-unknown-elf-clang --target=riscv32-unknown-elf -march=rv32i -c -o I-ENDIANESS-01.o -I ../../../riscv-test-env -I../../../riscv-test-env/p -I ../../../riscv-target/ri5cy I-ENDIANESS-01.S
```
Amongst other errors (see issue #105), this failed with
```
I-ENDIANESS-01.S:40:561: error: operand must be a valid system register
name or an integer in the range [0, 4095]
.section .text.init; .align 6; .weak stvec_handler; .weak mtvec_handler;
.globl _start; _start: j reset_vector; .align 2; trap_vector: csrr t5,
mcause; li t6, 0x8; beq t5, t6, write_tohost; li t6, 0x9; beq t5, t6,
write_tohost; li t6, 0xb; beq t5, t6, write_tohost; la t5,
mtvec_handler; beqz t5, 1f; jr t5; 1: csrr t5, mcause; bgez t5,
handle_exception; j other_exception; handle_exception: other_exception:
1: ori gp, gp, 1337; write_tohost: sw gp, tohost, t5; j write_tohost;
reset_vector: csrr a0, mhartid; 1: bnez a0, 1b; la t0, 1f; csrw mtvec,
t0; csrwi sptbr, 0; .align 2; 1:; la t0, 1f; csrw mtvec, t0; li t0, -1;
csrw pmpaddr0, t0; li t0, 0x18 | 0x01 | 0x02 | 0x04; csrw pmpcfg0, t0;
.align 2; 1:; la t0, 1f; csrw mtvec, t0; csrwi medeleg, 0; csrwi
mideleg, 0; csrwi mie, 0; .align 2; 1:; li gp, 0; la t0, trap_vector;
csrw mtvec, t0; li a0, 1; slli a0, a0, 31; bltz a0, 1f; fence; li gp, 1;
; ecall; 1:; la t0, stvec_handler; beqz t0, 1f; csrw stvec, t0; li t0,
(1 << 0xd) | (1 << 0xf) | (1 << 0xc) | (1 << 0x0) | (1 << 0x8) | (1 <<
0x3); csrw medeleg, t0; csrr t1, medeleg; bne t0, t1, other_exception;
1: csrwi mstatus, 0; init; ; ; la t0, 1f; csrw mepc, t0; csrr a0,
mhartid; mret; 1: begin_testcode:
```
Unlike GCC, Clang does not know about the sptbr (Supervisor Page-Table Base Register) CSR - part of the draft RISC-V Privileged spec, which was renamed to `satp` in the ratified spec. This code comes from an expansion of one of the macros, `INIT_SPTBR` in `riscv-test-env/p/riscv_test.h`

allenjbaum, 2020-04-29:
in addition to 
riscv-test-env/p/riscv_test.h,        sptbr is mentioned in these other files:
riscv-test-env/v/vm.c <--uses SPTBR_MODE, SPTBR_MODE_SV32, SPTBR_MODE_SV39, 
               but I don't know where they are defined
riscv-target/sifive-formal/formalspec-env/p/riscv_test.h
riscv-test-suite/deprecated/rv32mi/rv64mi/illegal.S
riscv-test-suite/deprecated/rv32mi/rv64si/dirty.S
riscv-test-suite/deprecated/rv32si/rv64si/dirty.S

This should be fixable by simply replacing references to sptbr with satp. We could further change the name of the INIT_SPTBR macro to INIT_SATP and aliasing INIT_SPTBR just in case.
I don't know what to do about the SPTBR_MODE* values, because they aren't defined in the compliance repository.

allenjbaum, 2021-07-15:
We will not fix the references in the deprecated tests.
I believe support for the sifive-formal simulator has been dropped - but eeen if not, they own that code, and target directories are going away (as are  the test-env directories when we swithcover to riscof,  so I will close this in 3 weeks unless there are objections

jrtc27, 2021-07-18:
LLVM lets you write sptbr these days, though it will unconditionally print a warning that you're using a deprecated CSR name and should fix your code.

closed 2021-08-17 by allenjbaum

redpanda3, 2025-04-06:
Same problem here.