riscv/riscv-arch-test #108: RI5CY's `compliance_io.h` fails to compile with clang

bluewww, 2020-04-22:
Errors with
```
riscv32-unknown-elf-clang --target=riscv32-unknown-elf -march=rv32i -c
-o I-ENDIANESS-01.o -I ../../../riscv-test-env -I
../../../riscv-test-env/p -I ../../../riscv-target/ri5cy I-ENDIANESS-01.S

../../../riscv-target/ri5cy/compliance_io.h:187:12: error: operand must
be a bare symbol name
    la t3, 0x10000000; sw a0, (0)(t3);;
```
because strictly speaking `la` doesn't accept a constant.

See also https://github.com/riscv/riscv-compliance/issues/105

jeremybennett, 2020-04-22:
@bluewww Are you OK to fix this?

allenjbaum, 2020-04-23:
The simplest fix is change a single character, replacing "la" with "li".

jeremybennett, 2020-04-24:
> The simplest fix is change a single character, replacing "la" with "li".

`li` is a pseudo instruction which translates in many ways (which are not documented in the spec). I think what is needed is:
```asm
        auipc t3, 0x10000
```
If the constant is hash defined, then it might have to be
```asm
        auipc   t3, (CONST >> 12)
        addi    t3, t3, (CONST & 0xfff)
```
although with the particular constant value here, the second instruction would be a no-op.

bluewww, 2020-04-28:
Looked into this, but currently I'm hitting another problem:
```bash
export RISCV_PREFIX=riscv32-unknown-elf-
export VSIM_WORK=/home/bluew/projects/riscv/tb/core/work/
export TARGET_SIM=/home/bluew/projects/riscv/tb/core/testbench_verilator
export RISCV_DEVICE=rv32imc
export RISCV_TARGET=ri5cy
make RISCV_ISA=rv32i && make RISCV_ISA=rv32im && make RISCV_ISA=rv32imc
```
results in
```bash
make  \
        RISCV_TARGET=ri5cy \
        RISCV_DEVICE=rv32imc \
        RISCV_PREFIX=riscv32-unknown-elf- \
        run -C /home/bluew/tmp/riscv-compliance/riscv-test-suite/rv32i
make[1]: Entering directory `/home/bluew/tmp/riscv-compliance/riscv-test-suite/rv32i'
Compile /home/bluew/tmp/riscv-compliance/work/rv32i/I-ENDIANESS-01.elf
/home/bluew/.riscv/lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot open output file /home/bluew/tmp/riscv-compliance/work/rv32i//home/bluew/tmp/riscv-compliance/work/rv32i/I-ENDIANESS-01.elf: No such file or directory
collect2: error: ld returned 1 exit status
/bin/sh: /home/bluew/tmp/riscv-compliance/work/rv32i//home/bluew/tmp/riscv-compliance/work/rv32i/I-ENDIANESS-01.elf.objdump: No such file or directory
/bin/sh: /home/bluew/tmp/riscv-compliance/work/rv32i//home/bluew/tmp/riscv-compliance/work/rv32i/I-ENDIANESS-01.elf.readelf: No such file or directory
/bin/sh: /home/bluew/tmp/riscv-compliance/work/rv32i//home/bluew/tmp/riscv-compliance/work/rv32i/I-ENDIANESS-01.elf.nm: No such file or directory
riscv32-unknown-elf-objcopy: '/home/bluew/tmp/riscv-compliance/work/rv32i//home/bluew/tmp/riscv-compliance/work/rv32i/I-ENDIANESS-01.elf': No such file
make[1]: *** [/home/bluew/tmp/riscv-compliance/work/rv32i/I-ENDIANESS-01.elf] Error 1
make[1]: Leaving directory `/home/bluew/tmp/riscv-compliance/riscv-test-suite/rv32i'
make: *** [simulate] Error 2
```

The following commit introduced the problem:
```
fba4324e62a85bd5de5c73d58d1ebd8edfb00da9 is the first bad commit
commit fba4324e62a85bd5de5c73d58d1ebd8edfb00da9
Author: eroom1966 <moore@imperas.com>
Date:   Mon Feb 17 10:03:59 2020 +0000

    Add Parallel make for supported targets
```

allenjbaum, 2020-04-28:
The issue may be that not all OSes have support for parallel make (centos
does not).
I don't know how to disable it; riscof had a similar issue, which is
supposed to have been fixed, though I don't know the details.
The only other target that explicitly defines LOCAL_IO_PUTC is riscvOVPsim,
and it's defines as
    .word RVTEST_CUSTOM1,    where RVTEST_CUSTOM1 is a #define of a hex
constant which translates to LW X0, 0(A0) - an unofficial noop
riscvOVPsim must interpret that specially to output the byte in A0.

I'm guessing ri5cy interprets a store to 0x10000000 as an output to an
interpreted UART
So, I believe the correct, simplest fix is to simply change the LA to LI in
the #define LOCAL_IO_PUTC(_R)

bluewww, 2020-04-29:
I don't believe this is a parallel make problem. I can't get it to run on opensuse tumbleweed either. It errors with the same message

neelgala, 2020-04-29:
@bluewww 
Yes you are right, that particular issue is because of changes in the root-makefile (possibly to support parallel make). You will have get rid of the $work_disr_isa.. I had to do these changes for spike/riscvovpsim in my last PR.

In my PR: https://github.com/riscv/riscv-compliance/pull/101/files 
Look for changes in riscv-target/spike/device/*/Makefile.include and you'll know what to do.

eroom1966, 2020-04-29:
To prove the issue is related to parallel build, this can be disabled by

make PARALLEL=0 ...

I suspect there is an intermediate file overwritten by multiple concurrent threads

allenjbaum, 2020-05-27:
Has the experiment of make PARALLEL=0 been tried?
IF it works - is this a bug that can be fixed globally, or is a strictly ri5cy makefile that needs to be updated.
Do we know if any other models or environments break?

allenjbaum, 2020-11-30:
@bluewww 
IF this isn't getting fixed, isn't being closed, and only being reported on a ri5cy supplied filed,  I'm going to have ri5cy support removed from the repo in 2 weeks from today, - so this will be closed one way or another.
This has been hanging on far too long.

bluewww, 2020-11-30:
@allenjbaum 
2h hours ago @MaskRay reported that llvm9 fixed one part of this issue. See https://github.com/riscv/riscv-compliance/issues/105#issuecomment-735545774
So it seems like llvm agrees that their assembler needs fixing and not this piece of code.

Regarding the other bug (`make PARALLEL=1` regression) I give it a look this week.

bluewww, 2020-12-03:
Here is the fix anyway.

bluewww, 2020-12-05:
Ok it's strictly a clang version issue and it seems like this will be resolved by specifying a minimum required clang version.

closed 2020-12-05 by bluewww

allenjbaum, 2020-12-05:
Ah, OK, it looked like it was simple enough to go ahead and fix for us. Are
you the ri5cy model maintainer? IF so, we need to talk off of github.

bluewww, 2020-12-05:
Yes. I added support for ri5cy. I sent you an e-mail.