riscv/riscv-arch-test #109: Release v0.2

olofk, 2020-05-08:
I'm maintaining out-of-tree riscv-compliance target code for two RISC-V implementations (SERV and SweRV). I had some bug reports coming in recently that made me realize that the current master had an API change (in the guise of a commit called Add Parallel make for supported targets) since v0.1

It would be great to have a v0.2 release so that I can point my users to that for a known good vetsion instead of a random git commit

P.S. I'm seeing hints of something called riscof framework. Is this something I should be aware of?

neelgala, 2020-05-09:
So I think what the issue is: since the addition of the parallel make api, there have also been upgrades to the test. So if you are using an older version, its quite possible that some bugs aren't being detected. 

That said, I think @eroom1966 @simon5656 can suggest how to proceed on this

RISCOF is the next version of compliance  - possibly v0.2.
Details of riscof: https://gitlab.com/incoresemi/riscof
Details of riscv-config:  https://github.com/riscv/riscv-config

PS: If you are sure its the paralle make, then the fix is in the arguments of your target Makefile.include files. Refer to PR: https://github.com/riscv/riscv-compliance/pull/101/files#diff-0be0f8dee7d8097a0906e624fefe1ec0 for possible hints on changes required.

olofk, 2020-05-09:
Thanks for the links. Will read up on the new framework and look into adding support for my targets

I would say this is all the more reason to do a new release, if the current framework is in maintenance mode. Just slap on a tag, write a couple of lines in a release note and use the release to communicate to others that nothing more will happen with this code base and that people should turn their eyes towards the riscof framework in the future.

And yes, I'm sure it was the parallel make that broke my target files. I already fixed it for one target. But I must also take the opportunity to say that I'm not impressed by how these changes were communicated. The name of the commit didn't indicate ant API changes and it seems only ovpsim target was fixed which broke the other targets

simon5656, 2020-05-09:
Hi Olof

Thanks for the feedback - I am sorry you had a problem with the way changes were communicated regarding the move to parallel and the fact that riscvOVPsim target was fixed.

Can you suggest how we could improve this process as clearly we don't have a good process - each company that has their own target has to be the one to move their target forward to match the evolution of the compliance framework - as many targets are not public and are specific to one company? The group does not require each target to made public and available for regression testing with CI. We tested 3 targets and went forward with the commit.

You are incorrect in your assumption that the current framework is in maintenance mode - it is evolving with the needs of the compliance challenge (e.g. going to parallel as their over 5,000 tests in the under development vector suite that we are developing) and it will evolve further.

The working group has a specification document regarding future direction of the framework, and there is still much discussion and no decisions as what it will be and what any plans and roadmap are for developing a new framework - and yes the riscof python code is an example that is talked about.

So, Olof - please suggest how we could have handled the parallel addition so that it did not rely on each simulator/target maintainer updating their specific target scripts.

Simon (Imperas)

olofk, 2020-05-11:
Hi Simon,

First of all, thanks everyone for all the work you put into the compliance tests. I'm a heavy user of them and have many times cursed how thorough they are when I make changes to my CPU that really shouldn't break anything :)

I *think* the particular issue with the parallel make API could had been solved in a backwards-compatible way by defining `$work_dir_isa` on targets that hadn't yet opted in to use the parallel make. But I might very well be wrong, and it's really beside the point. Stuff needs to break sometimes. I also hadn't updated my local copy of the repo in over a year. And I totally understand it's a hard issue since it's not feasible, or even possible to test all the targets in this repository by someone making changes. And it's of course even harder with targets that reside outside of this repo. I very much appreciate the possibility to have the target code externally, so that's something I would like to keep see support for (not least because I wrote the [patch](https://github.com/riscv/riscv-compliance/commit/6e46969fdaab38805b34e09068e37facc7d4ec13) to enable that :))

So let's focus on the process. I got three proposals that might improve the situation.

1. I think it's easier to communicate API breaks between versions, than between random git commits. So doing more frequent releases would help my target files, since I can say it works with riscv-compliance e.g. v0.4 up to v1.2.

2. I think it would be great also to have an opt-in mailing list or other forum where target maintainers can be notified of breaking changes and be given a chance to comment. This is a bit similar to how changes that affect multiple CPU architectures work in the Linux kernel

3. Make sure to communicate backwards-incompatible changes in the commit message and/or ChangeLog. Sometimes things break by mistake, and that's fine. But if they are known, they should be written down

I'm still a bit confused about the relation between riscv-compliance the the riscof framework. I noticed it mainly because there were a lot of bugs that were marked with `Fixed in RISCOF` which led to my conclusion that it was a new framework replacing the other one. But this issue report has already wandered far from its original purpose so I'll bring up those questions in another forum instead.  And feel free to close the issue if you're not planning on making a new release, which was the original issue I had

allenjbaum, 2020-05-11:
I'll comment on the last question about riscof.
When I was made chair of compliance, there had been a decision made that
tests were going to move from a static signature type of test to a
comparison against a formal model. I wasn't there when the decision was
made, but it was pretty clear to me that, once we started aggressively
testing privileged modes, we needed a framework that could describe the
actual implementation and select (or configure) tests that applied to those
modes.  Another goal (of mine) was to make tests easier to write or
generate. The 1.2.5 test format spec was the outcome of that,

Many of the open issues marked "fixed in riscof" are tests that simply
can't be fixed - but must be rewritten (or must have a separate test
written) because they make assumptions about exactly which set of
architectural options that a target has implemented. We can't simply allow
an implementor to decide which tests they want to run, and then declaring
themselves compliant with whatever subset they decided,  So, hey can't
simply delete the tests that don't pass from the list and insert one that
does; they need to declare what they implement and have the framework
select the tests.

 Riscof is a prototype (and the only one) that was written to take all of
that into account. The riscv-configuration repository was spun off from
that, since it had much wider applicability than just compliance.

My hope is that as soon as riscof becomes really production worthy, it will
be the next version of the framework, and will permit  testing far more
corners of the architecture than is possible now. There's a lot of work
that needs to be done there, including rewriting tests to meet the v1.2.5
spec (which can be done even before switching to riscof).

Note that there is another dependency that will take a while to resolve:
the formal model has to support all the possible options that the
architecture allows, and that is probably a ways off.

eroom1966, 2020-05-12:
@olofk 

Parallel build can be disabled by adding the make variable
PARALLEL=0
(but the default is set as PARALLEL=1)

I have not seen your code, but if copied from one of the other targets, a likely error is the '**temp**' file used in one of the commands, for example here is an example form the spike script

```
RUN_TARGET=\
    $(TARGET_SIM) $(TARGET_FLAGS) --isa=rv32imc \
        +signature=$(work_dir_isa)/$(*).signature.output \
        $(work_dir_isa)/$< 2> $(work_dir_isa)/$@; \
                                cat $(work_dir_isa)/$(*).signature.output | sed 's/.\{8\}/& /g' | \
                                awk '{print $$4 " " $$3 " " $$2 " " $$1}' | sed 's/ /\n/g' > temp; \
                                mv temp $(work_dir_isa)/$(*).signature.output;
```

so what happens here is that multiple parallel processes all write to the same file - not good
this should be qualified as an intermidiate file by using the stem, eg

```
RUN_TARGET=\
    $(TARGET_SIM) $(TARGET_FLAGS) --isa=rv32imc \
        +signature=$(work_dir_isa)/$(*).signature.output \
        $(work_dir_isa)/$< 2> $(work_dir_isa)/$@; \
                                cat $(work_dir_isa)/$(*).signature.output | sed 's/.\{8\}/& /g' | \
                                awk '{print $$4 " " $$3 " " $$2 " " $$1}' | sed 's/ /\n/g' > $(*).temp; \
                                mv $(*).temp $(work_dir_isa)/$(*).signature.output;
```

Thx
Lee

allenjbaum, 2020-12-05:
IF the above suggestions have fixed the issue, could you close it please?

FYI: there is another update coming very shortly that will have some further changes (see PR#154)
IT will have a LOT of new tests whose format conforms to the test format spec, a standard trap handler, streamlined test directory, a more regular naming convention, and updated documentation.
There is a migration document that has details on what to change (not very much, as there are aliases and symlinks for backwards compatibility, but there are a few macros that need renaming).

We would appreciate it if you could try it out and let us know what breaks.

olofk, 2021-04-07:
I was just reminded of the existence of this bug report and don't think there is any point in keeping it open

closed 2021-04-07 by olofk