riscv/riscv-arch-test #110: adding make target to define whether assertions are on or off

Imperas, 2020-05-18:
When creating new tests the process is
1) create new test/tests
2) ensure it/they assembles
3) run on a reference - to check the reference is setup
4) turn on assertions - to ensure that the self checking part of the test is passing
5) turn off assertions and run again - creating the signature
6) copy the signature into the references

Today - the turning on and off of the assertions is done by editing the devices compliance_io.h

Can this please be made a make target to # define to turn on/off assertions

thanks

eroom1966, 2020-05-27:
I have put a fix for a make command line variable to enable the assertions #112

simon5656, 2020-05-27:
closing as make now has a target to turn on/off asserts

closed 2020-05-27 by simon5656