riscv/riscv-arch-test #114: riscv-tests vs riscv-compliance

T3KT4K, 2020-06-03:
What are the differences in the goals of the two projects? It seems there is some overlap so I am slightly confused.

jeremybennett, 2020-06-03:
This is the repository of the RISC-V compliance task group. It aims to provide a set of tests which any implementation must pass to be certified as compliant with the RISC-V specification.  Note that this is not comprehensive verification of a design, just sufficient to give confidence that the implementation meets the specification.

riscv-tests is a repository of unit tests for RISC-V processors, largely maintained by the SiFive engineering team. It includes tests of things like semihosting, which are about the software architecture built on top of the RISC-V hardware.

HTH

T3KT4K, 2020-06-03:
Understood. Thanks

closed 2020-06-03 by T3KT4K