riscv/riscv-arch-test #115: How to support on-board execution？

adchd, 2020-06-05:
At present, the test framework or case is executed on the simulator. I wonder if it can support the execution on the board? （Such as executed on Linux operating system）

stnolting, 2020-08-04:
You can execute the tests on a board - so on actual hardware. For each test scenario you will need to program the board with the actual test, execute it and somehow get the results back to the host (maybe via UART) to compare against the references.

allenjbaum, 2020-08-04:
Can anyone who has actually run compliance tests on hardware (including an
FPGA implementation) speak up and show how you did it?
This would be a good case study. The lack of any response would also be an
interesting, and negative, case study.

allenjbaum, 2020-08-06:
I have been reminded that there was a competition about 18 months ago where multiple FPGA prototypes were designed and passed compliance. see https://riscv.org/2018/12/risc-v-softcpu-contest-winners-demonstrate-cutting-edge-risc-v-implementations-for-fpgas/

neelgala, 2020-08-06:
I haven't tried the current framework on FPGAs. However, RISCOF does have a gdb plugin running over spike : https://gitlab.com/incoresemi/riscof-plugins/-/tree/master/gdb

If you don't have debug support on your device then as mentioned by @stnolting you will have define your own mechanism to read out the signature at the end of the test and dump it somewhere on the host system. I hope the device macros like halt would suffice for this.

adchd, 2020-08-10:
> I haven't tried the current framework on FPGAs. However, RISCOF does have a gdb plugin running over spike : https://gitlab.com/incoresemi/riscof-plugins/-/tree/master/gdb
>
> If you don't have debug [...]

I only tried the current framework (riscv-compliance) on FPGAs . It's can be run but need to change code. 
I am trying run on FPGAs  based on riscof framework .

bnossum, 2020-09-23:
A follow up on the request by @allenjbaum on Aug 5, I run compliance tests on my icebreaker board via an usart, testing my riscv implementation (midgetv). See [here](https://github.com/bnossum/midgetv/blob/master/work/compliance/README.md).

allenjbaum, 2021-04-21:
Can this issue be closed now? Note that riscv-compliance has been renamed riscv-arch-test, so scripts may need to be updated to account for that. Tests still assume specific architectural options for some tests (e.g. c-extension and misaligned handling) which will be resolved in the next fversion of the framework.

dvc94ch, 2021-06-12:
I was wondering if there was a more structured way to get the test cases than regexes. Would be useful to be able to extract the comments above the test cases like for example here [0] into a csv file (pre/post conditions) to be able to more easily use them in custom test suites or runners. The yaml used to generate those is fairly complex and high level. Would also be nice to include the actual instruction being tested in hex format.

- [0] https://github.com/riscv/riscv-arch-test/blob/master/riscv-test-suite/rv32i_m/I/src/add-01.S#L51

indumathiraju, 2022-04-06:
is it support for u74 unmatched board

allenjbaum, 2022-04-06:
In theory, yes.
Whoever is performing the tests need to implement the model-specific macro
support, as defined in the architectural tests spec
(any macro prefixed with MODELTEST_ string) but basically,
 -  the ability to download ELF formatted tests into onboard memory
 - the ability to start tests running at the start address in those ELF
files
 - the ability to recognize when the board tells you that the test is
complete (generally an MMIO write to a specific address)
 - the ability to dump a specific area of memory into a text file in a
simple hex format
There is a note in issue #115 where someone has demonstrated it running on
real hardware
(as opposed to a RTL model) so it can be done, but you'll need to ask
someone at SiFive if they've actually done that work,.

bnossum, 2022-04-07:
I am probably the "someone" that demonstrated it running on real hardware. I just want to point out that my testing was done 18 months ago, and the macros of the testsuite has been changed since then. The principle is, however, still sound.

allenjbaum, 2022-04-07:
I didn't think required model specific macros have changed at all.
Optional macros (signature updates, for sure) have changed, but should be
backward compatible.
But I do agree that the principle is still sound.
Thanks for chiming in.

If you have any documentation about how you did that could share, we'll add
it to the repo.

allenjbaum, 2022-04-23:
As I believe this question has been answered, I'm closing this issue. If the answers are not satisfactory, reopen.

closed 2022-04-23 by allenjbaum

MarcKarasek, 2022-10-11:
The test suite is really ment to test against the ISA prior to tape out.
If you are testing after that and find a bug/problem in the chip...  Not
good

allenjbaum, 2022-10-11:
Well yes - it means you haven't done proper DV.
DV is for catching bugs.
Arch-tests are for catching misreadings, misinterpretations, or just plain
ignoring details of the spec

allenjbaum, 2022-10-11:
Oh, and a correction to my earlier email, where I said
 (any macro prefixed with MODELTEST_ string)
but should have said :
    (any macro prefixed with RVMODEL_ string instead of RVTEST_ string)