riscv/riscv-arch-test #116: loss of 64bit test infrastucture

simon5656, 2020-06-05:
@neelgala 
in this commit https://github.com/riscv/riscv-compliance/commit/a0246089194b9a98dbbd61994346d0415390d0fb
youmoved the 64bit tests into wip
what did you do with the target device code? can you please also move that back to wherever it should be
for example you have removed riscv-target/riscvOVPsim/device/rv64i and rv64im directories, they should not have been deleted, but also should be be moved to wip sub dirs

pls reinstate things 
so that when people want to run the 64bit tests - they just need to cp -r .. the dirs and they can use the tests.
thanks
Simon

neelgala, 2020-06-05:
Well @eroom1966 had approved the [pull request](https://github.com/riscv/riscv-compliance/pull/101) (twice infact with the quote "I am happy for these changes" ) .. anyways... I'll do a PR for this.

neelgala, 2020-06-05:
@simon5656 @eroom1966 I have created a branch: [revert_rv64_ovpsim](https://github.com/riscv/riscv-compliance/tree/revert_rv64_ovpsim) on the repo. 
Pulled in your old target file and updated the include files GCC_OPTS as well.

You can choose to re-order them in the fashion you prefer and make a PR once you are
done.

allenjbaum, 2020-08-06:
IS anyone looking into this? It should either get implemented or closed.

neelgala, 2020-08-06:
the branch already exists on repo. Simon or Lee should be able to edit it according to their needs and merge it.

allenjbaum, 2021-04-21:
Can this be closed now?

allenjbaum, 2021-04-26:
I'm going to close this unless someone tells me that this isn't working in the next week.

neelgala, 2021-04-28:
we should probably also get rid of the branch : revert_rv64_ovpsim if this is closed.

allenjbaum, 2021-04-29:
I actually don't remember why that branch was there; I thought had to do
more with target files.

Imperas, 2021-05-05:
Yes please delete branch and close.
Simon

allenjbaum, 2021-05-06:
deleted and closed

closed 2021-05-06 by allenjbaum