riscv/riscv-arch-test #117: Cannot build sail to be used as riscv-compliance target

Imperas, 2020-06-15:
@pmundkur 
Hi Prashant, 
are you the owner for the sail integration into the compliance framework ?
could I ask you to take a look at this issue I have raised on sail https://github.com/rems-project/sail-riscv/issues/59
It appears to generate a syntax error, but this is building from the Master branch
Is the Master branch the correct version to use, or is there a release ?

Thx
Lee

pmundkur, 2020-06-15:
You should be able to use the master branch of sail-riscv.  It looks like you are using an outdated sail compiler (version 0.10).  The current version in opam is 0.13.

allenjbaum, 2020-07-09:
IF this answer is correct, could you please close the issue?

eroom1966, 2020-07-09:
I managed to resolve this - thanks

closed 2020-07-09 by eroom1966