riscv/riscv-arch-test #121: Problem with base ISA EBREAK test's trap handler - no support for compressed instructions

stnolting, 2020-07-03:
I am not sure, if this is a real and intended scenario, but if you compile the rv32i base ISA tests (`RISCV_DEVICE=rv32i`) also using compressed instructions (e.g. `RISCV_ISA=rv32imc` -> using `-march=rv32imc` for the toolchain), the `ebreak` instruction test (`I-EBREAK-01.S`) will always fail.

The trap handler in this test program always assumes 32-bit aligned instructions. In case of the compressed `ebreak` instruction, the exception return address is not computed correctly.

    csrr  x30, mepc
    addi  x30, x30, 4  // this is wrong if the exception-causing instruction is compressed
    csrw  mepc, x30

This could be fixed using bit 1 of the RISC-V `mtinst` CSR (the bit and the CSR are compliant to the RISC-V privileged architecture specification):

    /* computing the return address in the trap handler (for compressed & uncompressed instructions) */
    csrr  x30, mepc
    
    csrr  x31, mtinst
    andi  x31, x31, 2   // isolate compression flag (bit #1): 0=compressed, 1=uncompressed
    
    addi  x30, x30, +2  // compressed instruction
    add   x30, x30, x31 // add another +2 (making +4) if mtinst inidicates uncompressed instruction
    
    csrw mepc, x30

allenjbaum, 2020-07-03:
Actually, the easy/correct fix for this to make sure the test (actually all
tests that are not in the rv32imc subdirectory) is compiled with
*.option norvc*. (and I notice that there is no C.EBREAK test in that
subdirectory)

But, the mtinst CSR does not exist in the current ratified v1.11 spec
And, even if it did, the version 1.12 draft from jun11 only defines mtinst
to exist if the hypervisor extension is present.
And even if the hypervisor extension is present, implementations are not
required to set any bit in mtinst.
So, the method you propose cannot be used to fix this (though it should not
need fixing if *.option norvc* has been used).

However, if you did need to fix it, MEPC can be used to load the opcode,
the code can look at the 2 LSBs to determine whether it is a compressed op,
and increment by 2 rather than 4 if it is.

Future versions of the compliance suite will always return from traps to a
4 byte boundary, with at least one halfword of padding between the
excepting instruction and the return point. This gives compliance code a
way to detect and handle compliance failures that don't trap when they
should.

Having said all that: this is a more general bug than just EBREAK, since
there doesn't seem to be any  *.option norvc* present anywhere in the
repository (outside of deprecated tests).

stnolting, 2020-07-05:
Thanks for the clarification!

closed 2020-07-05 by stnolting