riscv/riscv-arch-test #122: Why riscv-compliance programs are not self checking?

davidmlw, 2020-07-08:
I noticed that riscv-compliance programs dump a golden out for reference. Why not are they not self checking and output 1-bit passed or failed? Are there some specific considerations?

allenjbaum, 2020-07-08:
The original tests were in fact self checking. The problem is that the
Risc-V architecture has an overwhelming number of optional features and
configurations, and each of those potentially results in different test
results.

We need to be able to generate the correct result to check against even for
a self-checking test, and to ensure that the correct result is the one that
is being checked, we have to run the test with the golden reference model.

That golden model must be configurable to be able to handle all possible
legal  options and configurations.
So, by checking against the golden model, which would be
automatically configured by a YAML -encoded configuration file, we should
be able to test completely arbitrary configurations, and we don't have to
figure out what the "right" answer is if we add new tests to the existing
test suite; it will be generated automatically.

You won't see this capability usedmuch for ops in the base ISA, but when we
get to privilege level tests and hypervisor tests, this will be used
extensively.

davidmlw, 2020-07-08:
I see. Nice design. Then we need a way to integrate the OVP into the verification environment for a complex and maybe dynamic test. Are there some examples that combine OVP and a (System)Verilog design?

allenjbaum, 2020-07-08:
The next version of the framework (not the current version) will allow
comparisons between any two models.
OVPSim could be one, and a system verilog simulator could be another.

Note that the golden reference model, written in Sail is the official
simulator, and that is the one that must be use to license use of the
Risc-V trademark

simon5656, 2020-07-08:
Hi Liwei

Yes - over the last 10 years we have had many people using Imperas OVP reference models integrated in their hardware DV environments to validate their designs.

In the RISC-V world Imperas have an integration of the OVP simulators in SystemVerilog in use with several customers.

Recently Imperas were featured by OpenHW in the TV show about our SystemVerilog integration. The video link is here with it starting at the section of interest: https://www.youtube.com/watch?v=GNElOjxo0GQ&feature=youtu.be&t=1254

You can download their GitHub to have a look:  https://github.com/openhwgroup/core-v-verif/tree/master/vendor_lib/imperas/riscv_CV32E40P_OVPsim

And you can download an example of RISC-V OVP reference model in SystemVerilog  from OVPworld (when registered):  search for 'Examples_SystemVerilog' in page https://www.ovpworld.org/dlp/

thanks
Simon

allenjbaum, 2020-07-09:
IF you're satisfied with these answers, please close the issue

closed 2020-07-09 by davidmlw