riscv/riscv-arch-test #123: How to detect a test is finished? It'd be better to have device register map and protocol spec.

davidmlw, 2020-07-08:
As the current test suite is not a self checking one, and it dumps the signatures to `tohost` address. How to detect a test is finished? Is there another tohost address to indicate the current test suite is finished so the env will check the received signatures?

If we take the `host` as a device, it is better to have a register map specification. For example, starting and finishing a test suite, case, and a beating protocol if the tests take too long to run.

davidmlw, 2020-07-08:
It seems that it could be implemented by the env implementor oneself by defining `RV_COMPLIANCE_CODE_END`

allenjbaum, 2020-07-08:
Close, but not quite correct.
The  RV_COMPLIANCE_HALT macro is model specific, and indicates that the
test is completed.
Note that there can be more than one instance of this in a test, e.g. at
both sides of a branch.

allenjbaum, 2020-07-09:
If you are satisfied with this answer, please close the issue

closed 2020-07-09 by davidmlw