riscv/riscv-arch-test #125: Request to stop hosting closed source code on riscv repo

ShashankVM, 2020-07-16:
Because RISC-V is a project of The Linux Foundation, I request you to stop hosting closed source code on your repo.
The Linux Foundation supports only open-source projects.
According to https://opensource.guide/legal you cannot use closed-source code in your project. The riscvOVPsim simulator is clearly closed source.
It is meaningless to say that the models are open-source as the models cannot be used without the simulator. 
As an analogy, it would be like licensing some MATLAB code under GPL. And MATLAB code cannot be used without purchasing proprietary software from MathWorks.
I request you to stop hosting riscvOVPsim binary on this repo. You should also remove the OVP models.
Or else, this will be the first time in history that closed-source code is hosted by a project of The Linux Foundation.

simon5656, 2020-07-16:
Thank you for your question. I think you are a little confused by GPL and binary programs. What you are saying with your matlab example relates to the use of GPL with commercial products and I agree with https://opensource.guide/legal that if a binary makes use of GPL it needs to be provided as source as well. There is no issue with GitHub accounts hosting free binaries and there is no issue with the riscvOVPsim binary as it includes no GPL code. For further information, I suggest you study https://www.gnu.org/licenses/gpl-3.0.en.html to see how GPL relates to binaries. Thanks. Simon.

ShashankVM, 2020-07-16:
I'm not specifically talking about GPL. My concern in that riscv-compliance project has a dependency on closed source software. This is not allowed according to Section 8 of https://opensource.guide/legal

allenjbaum, 2020-07-16:
There is no dependency on using OVPsim for compliance, now or in the future.
Strictly speaking, the only model that is required for compliance is the Sail formal model. All the others are there only for convenience and are completely optional.

-Allen

ShashankVM, 2020-07-16:
@allenjbaum 
Yes, I agree that OVPsim may not be necessary for software engineers who want to build software compliant with RISC-V.
Suppose a hardware developer is working on adding new features like bit manipulation to RISC-V. By referring the README.md of https://github.com/riscv/riscv-bitmanip it is neither obvious nor does it seem that OVPsim is optional

neelgala, 2020-07-16:
I agree with Allen and also partially with Shashank's arguments.
Since riscvOVPsim is not a golden reference model, it would make sense to treat it similar to other compliance targets like: spike, grift, etc. If not, then we might end up hosting all simulators and compliance targets on the compliance repo.

Just like other targets, ovpsim could include a README to build and use compliance with ovpsim.
Also the size of the compliance repo - which shuold be shipping only assembly files - is pretty high because of the ovpsim.

And moreover, the commit history is cluttered with ovpsim updates which actually have nothing to do with the compliance suite themselves.

ShashankVM, 2020-07-17:
@allenjbaum @simon5656 
The license for the riscv-compliance repo is also ambiguous and incomplete. https://github.com/riscv/riscv-compliance/blob/master/COPYING.BSD
I am skeptical that there can be an open-source license for the repo if risvOVPsim or the OVP models continues to be a part of it.

simon5656, 2020-07-17:
Thank you for the feedback - we will make things clearer in the license file.

neelgala, 2020-07-17:
let's keep this open until its resolved. That's the same protocol for other issues as well.

aswaterman, 2020-07-17:
Regardless, it doesn't make sense to host simulators within this repo.

simon5656, 2020-07-17:
OK lets discuss in next compliance working group meeting.

raulbehl, 2020-07-18:
How does having UVM based testbenches is different from using OVPSim? I understand that UVM is an IEEE standard and we `might` have an open-source simulator which supports UVM in the future. Given the current situation with commercial simulators only supporting (and driving) UVM, if there is an open source project (ibex as an example) which uses UVM for verification needs really doesn't help the open source committee. I find it very hard to believe for an individual to have access to the commercial simulators which support UVM.

allenjbaum, 2020-07-18:
IF the problem is something in the readme of the bitmanip extension,
shouldn't it get changed there if anywhere?

allenjbaum, 2020-07-20:
Please be very clear that compliance is NOT verification; it tests architectural features, while verification tests microarchitectural features as well - those may indeed need UVM, but that's not our problem.

Please be clear that IBEX may be open source, but it is not a Risc-V project; it is its own project and can use any tool it wants to use for verification, open or proprietary.
Any Implementor is always free to choose any tool - proprietary or not - to **verify** their design.
However, implementors must use Risc-V open source tools (the Sail formal model, the risc-v compliance framework, risc-v tools, etc) to test **compliance** of their design.

Implementors are free to use riscOVPSIM (a proprietary, but free-to-use tool) to compare their design.
RiscOVPsim is highly configurable, and easy to use.
But, even if the results of your implementation matches riscOVPsim, your implmentation is not considered compliant.
Only if it matches the Sail formal model results using the risc-v compliance framework and the risc-v compliance test suite is it considered compliant.

ShashankVM, 2020-07-21:
> Please be very clear that compliance is NOT verification; it tests architectural features, while verification tests microarchitectural features as well - those may indeed need UVM, but that's not [...]

@allenjbaum, Is Sail the only formal model to be used for testing compliance? Or can other formal models be used?

neelgala, 2020-07-21:
SAIL is the officially adopted formal model by the foundation. So to hand-out a compliance certificate (or something similar) the foundation expects the SAIL model to be used the reference model during compliance suite testing.

allenjbaum, 2020-07-21:
Correct, what Neil said.
Other formal models (or non-formal model) can be used to test your design,
but for Risc-V compliance (TM), Sail is the one that must be used in order
to be able to license the trademark.

allenjbaum, 2020-11-09:
I think this can be closed now.
RiscvOVPsim has been moved out of the riscv-compliance repository into a separate one: github.com/riscv-ovpsim
What remains is just a readme file that describes where to find it.

Separately, we are also considering removing the support files that have been included in the repository for various models and having them moved into the repositories of the models themselves. If changes are required to get those models to run for some reason, having in repositories that the models control is a much better place.  There may be exceptions for the models that we are required to run (Sail, spike, and QEMU) but even that is being examined.

The need for these model specific files will be reduced when we move to the v.2 framework in any case.

closed 2020-11-10 by ShashankVM