riscv/riscv-arch-test #130: How to redirect address of begin_signature

davidmlw, 2020-08-04:
As discussed in other threads, not all test cases in riscv-compliance are self-checking, some cases dump the execution results and depend on environment to check the signature results.

Is there an elegant way to redirect or specific the address of `begin_signature` in `link.ld` script? In this way, one can provide a virtual device to handle the execution results, rather than hack the main memory address bus.

https://github.com/riscv/riscv-compliance/blob/7cc155620865297512cdace3c696441e3977462a/riscv-test-env/p/riscv_test.h#L237-L251

stnolting, 2020-08-04:
I am using the `RV_COMPLIANCE_HALT` macro to copy the data from the signature region to a memory-mapped output device (at address 0xFFFFFF88). Maybe this can solve your problem.

```
#define RV_COMPLIANCE_HALT                                                    \
  	la a0, begin_signature;                                               \
	la a1, end_signature;                                                 \
        li a2, 0xFFFFFF88;                                                    \
        copy_loop:                                                            \
        beq a0, a1, copy_loop_end;                                            \
        lw t0, 0(a0);                                                         \
        sw t0, 0(a2);                                                         \
        addi a0, a0, 4;                                                       \
        j copy_loop;                                                          \
        copy_loop_end:                                                        \
        j copy_loop_end;                                                      \
        RVTEST_PASS                                                           \
```

davidmlw, 2020-08-05:
Nice solution! I was stucked by link script.

> I am using the `RV_COMPLIANCE_HALT` macro to copy the data from the signature region to a memory-mapped output device (at address 0xFFFFFF88). Maybe this can solve your problem.
>
> ```
> #define [...]

allenjbaum, 2020-08-06:
IF you have the answer you need, could you please close this issue?

closed 2020-08-07 by davidmlw