riscv/riscv-arch-test #131: Why reference_output has four extra results?

davidmlw, 2020-08-04:
from the asm source code, one can see that there are 32 (0-31) cases and will generated 32 results. But why the reference output has 4 extra results? 

https://github.com/riscv/riscv-compliance/blob/7cc155620865297512cdace3c696441e3977462a/riscv-test-suite/rv32i/src/I-ADD-01.S#L136-L137
https://github.com/riscv/riscv-compliance/blob/7cc155620865297512cdace3c696441e3977462a/riscv-test-suite/rv32i/references/I-ADD-01.reference_output#L31-L36

allenjbaum, 2020-08-04:
Having extra lines in the reference and in the signature area is OK- there's no harm. 
The extra values in the reference match the initial values of the signature, which is correct if they never get written. 

The real problem is that the signature area is 35 words, but the reference is 36 - so it's checking a word outside the  signature area. IF that just happens to be zero (the reference value)- no problem. But you can't guarantee that in all environments, so it should be fixed.

davidmlw, 2020-08-05:
I think I know the reason. The reference out is from `begin_signature` to `end_signature`.

See: https://github.com/riscv/riscv-compliance/issues/130#issuecomment-668511399

allenjbaum, 2020-08-05:
I would say that The reference out **should be** from begin_signature to end_signature.
A problem is that the length of reference out (36*4B) does not match the size of end_signature to begin_signature (35*4B). They are supposed to match. 
This is not a problem with just this test; I looked at a couple of other with exactly the same issue.
These tests will all be replaced and it seems to be a harmless bug, so I think this should be marked as "will not be fixed" and closed.

eroom1966, 2020-08-05:
I can tell you why this is the case.
Firstly the signature is determined by the start/end labels, and in this case the start end labels encapsulate 35 (4-byte) words

the spike signature output format works in 16 byte chunks, so blocks of 4 x 4-byte words
this means that the size of the reference must be aligned to a 16 byte (4 x 4-byte) boundary
Originally signature references were output in this 16 byte format, this was modified by @neelgala - under the direction of the committee, to produce 1 x 4-byte signature lines
https://github.com/riscv/riscv-compliance/pull/19

looking at the end of the ASM file there are 7 blocks of data each 5 words long, so this is 35 words in total
in order to match spikes signature output format, this requires rounding to a 4 word alignment, which is 36 words
hence the signature reference contains 36 4-byte words, rounded up from 35.

The signature labels in this case could be shortened, but it is no big deal, since the signature itself, is initialized as part of the elf definition

If the intent was to have non 4x4-byte alignment, then spike needs to change the way it produces its signature dump

I hope that clarifies the position

Thx
Lee

allenjbaum, 2020-08-05:
Ah, that explains how it happened.
As long as the framework compares (only) the signature from signature_start
to signature_end, and the reference size is >= the signature size, this
isn't a problem (and, of course, the test shouldn't store things outside
the signature....).
If the signature compare looks at just the dump size, and not the start to
end size, there could be a problem (false negative).
This will get weirder when we are comparing two different models that dump
different sizes; it sounds like spike should be updated, since the clear
intent is to only look at the signature area.
Good to know the history though - thanks.

allenjbaum, 2020-08-05:
If this is indeed harmless, can <someone> close this issue please?

Dracarys99, 2020-08-06:
In order to account for the different dump sizes of various simulators and prevent false negatives, the size of the signature region should also be multiples of 128/64 bytes. This should be one of the constraints mentioned in the Test format spec.

allenjbaum, 2020-08-06:
It's never been mentioned before so neither the spike nor any other similar
requirement are in the spec.
We can add a .fill to the RV_COMPLIANCE_DATA_END macro to force the size to
any specific size multiple, and could even add a .align to the
RV_COMPLIANCE_DATA_BEGIN macro to force alignment as well, in which case
the test spec doesn't need to change.

allenjbaum, 2020-08-06:
Actually, that's effectively how it is done here:

riscv-target/sifive-formal/formalspec-env/p/riscv_test.h
<https://github.com/riscv/riscv-compliance/blob/9f280717f26f50833357db9bfb77a8c79835f162/riscv-target/sifive-formal/formalspec-env/p/riscv_test.h>
   and here
riscv-test-env/p/riscv_test.h
<https://github.com/riscv/riscv-compliance/blob/9207bb3bab0de7cefbe02d816a7c8809c75abb81/riscv-test-env/p/riscv_test.h>
  though I think .fill would catch test errors that go past the end of the
area better.

neelgala, 2020-08-06:
using .align in GNU fills the section with zeros. Can we assume the same about other toolchains as well? If not then its a good guideline to indicate in the test-spec.

eroom1966, 2020-08-06:
> 
> 
> In order to account for the different dump sizes of various simulators and prevent false negatives, the size of the signature region should also be multiples of 128/64/32 bytes. This should be one of the constraints mentioned in the Test format spec.

Well .... possibly
If we were only saving XLEN length registers to the signature file - I would tend to agree
but there may well be data in the signature which is not extracted from a register, and does therefore not need to be aligned to the XLEN
For example if you support FD, then there are both single and double precision floating point registers

allenjbaum, 2020-08-06:
I'm not sure what the size of data in the signature has to do with either
the size of the signature area or its initial alignment; could you explain?
If we are saving Dfloat, there needs to be enough room in the signature
area. There may be a requirement to force alignment for that signature to
be 8B - but that has nothing to do with the total size of the signature
area being a multiple of some power of two, nor its initial alignment. The
signature area is an opaque array of 4B numbers, and no attempt is made to
decode their types.

Currently, the RV_COMPLIANCE_DATA_END macro can be defined by the model to
any alignment that a model requires; some models force it be >4B, some
don't care and don't force any alignment. We could augment the macros with
an optional alignment parameter, if that were important.

But, this may point out the need for the "Dummies Guide to Compliance Test
Integration Document" that is a tutorial for what an implementor has to do
to get their model to run the compliance framework. There is some
documentation for this, but it sounds like it doesn't go into enough detail.

closed 2020-08-07 by davidmlw