riscv/riscv-arch-test #132: Why not just use mepc for mret?

davidmlw, 2020-08-16:
This test case depends on mtval having been written the jumping target. While this is not a compliance assumption according to hte risc-v spec, it is: "When a trap is taken into M-mode, {\tt mtval} is either set to zero or written with exception-specific information to assist software in handling the trap." why not just use `mepc` for `mret`?

https://github.com/riscv/riscv-compliance/blob/7cc155620865297512cdace3c696441e3977462a/riscv-test-suite/rv32i/src/I-MISALIGN_JMP-01.S#L282-L284

https://github.com/riscv/riscv-isa-manual/blob/04604b2ab7cffcc276bdc0b7d7d2ff5f289eab9f/src/machine.tex#L2215-L2223

allenjbaum, 2020-08-16:
Meow points to the instruction room that trapped. In the case, it trapped because it branched to a misaligned address, if I recall correctly. So meet would return and simply get into an infinite loop continually retrapping..

-Allen

allenjbaum, 2020-08-16:
Sorry about autocorrect errors.

MEPC points to the instruction that trapped. In the case, it trapped because it branched to a misaligned address, if I recall the test correctly. So MRET would return to the faulting op and simply get into an infinite loop continually retrapping.

-Allen

davidmlw, 2020-08-17:
I see. We could use
```
 csrr    x30, mepc 
 addi    x30, x30, 2 
 csrw    mepc, x30 
```
The current test case depends on `mtval` holding the jumping target, which is not a mandatory behavior that a processor must implement.

allenjbaum, 2020-09-11:
The trap handler that is to included in future tests will always return to xEPC+8.
This takes into account all instruction sizes, all instruction alignments, and allows padding following tthe trapped up that can be use to insert code that detects that a trap did not occur when it should have (or that xEPC didn't have the correct value)
So, your code above is correct, and the value is 8. We do not adjust to the size of the instruction because we can't be guaranteed to be able to access and decode it (e.g. a PMP entry allows execute, but not read, privileges)

allenjbaum, 2021-04-21:
No reply after 6 months, so I'm proactively closing this

closed 2021-04-21 by allenjbaum