riscv/riscv-arch-test #133: Usage of pseudo instruction unimp

Imperas, 2020-08-19:
I have noticed that there is usage of the assembler instruction "unimp" in one of the header files
https://github.com/riscv/riscv-compliance/blob/master/riscv-test-env/p/riscv_test.h

This has been present since the first commit inherited from previous work.
There is no reference to 'unimp' in the unprivileged or privileged spec.
It appears the compiler produces the bit pattern 
` 414:   c0001073                unimp`

it is odd that the toolchain can produce a bit pattern that is not documented
should this be changed ?

Thx
Lee

aswaterman, 2020-08-19:
`unimp` is not a real instruction, but is rather a pseudoinstruction for `csrrw x0, cycle, x0` (which is always illegal since `cycle` is a read-only CSR).  Since it's just a pseudoinstruction and an assembler convention, it doesn't belong in the ISA spec, but it does arguably belong in the very incomplete ASM manual (https://github.com/riscv/riscv-asm-manual).

allenjbaum, 2020-08-19:
How is csrrw X0, cycle, X0 different than csrr X0, cycle  - which should be
legal?

pdonahue-ventana, 2020-08-19:
> How is csrrw X0, cycle, X0 different than csrr X0, cycle - which should be legal?

See table 9.1 in Zicsr.  `csrrw x0, cycle, x0` matches the first line which writes the CSR but does not read it.  `csrr` is an alias for `csrrs` per the (now removed) chapter 25.

allenjbaum, 2020-08-19:
Ah, I saw the statements about Rd=X0 "the instruction shall not read the
CSR and shall not cause any of the side effects that might occur on a CSR
read." and didn't read it carefully enough to see that did not apply to
Rs1=X0. Thanks.

allenjbaum, 2020-09-11:
So: can we close out this issue now? (and/or move it to the asm-manual?)

jeremybennett, 2020-09-11:
The original problem is still unresolved. However, a quick grep through the master branch shows there is only one offending file remaining, `riscv-target/sifive-formal/formalspec-env/p/riscv_test.h`

```c
//-----------------------------------------------------------------------
// End Macro
//-----------------------------------------------------------------------

#define RVTEST_CODE_END                                                 \
end_testcode:                                                           \
        unimp
```

Compliance should not be using any assembly language that is not standardized, since while it may currently be in GCC, there is no guarantee it will stay there, nor that it will be in other tool chains, neither open source nor proprietary. We need to work with what is in Chapter 25, table 25.2 of the ratified standard.

In this case, the fix is trivial, since it is only a macro and can be replaced by `csrrw x0, cycle, x0`.  Once this change is made, then the issue can be closed.

jeremybennett, 2020-09-12:
Thanks @debs-sifive for the quick fix. I've merged it, and so this issue can now be closed.

closed 2020-09-12 by jeremybennett