riscv/riscv-arch-test #135: Request for a Tag on this Repo

MikeOpenHWGroup, 2020-09-04:
Hi.  The OpenHW Group is using this repository for its compliance testing of the [cv32e40p](https://github.com/openhwgroup/cv32e40p) (and will almost certainly use it for [cva6](https://github.com/openhwgroup/cva6) as well).   It would be helpful to have a specific tag for us to use (much more people-friendly than the hash).

Thanks!

jeremybennett, 2020-09-04:
@MikeOpenHWGroup I've assigned to @neelgala who is one of the active maintainers. I note that so far the repository has not used tagging at all.

You might have to fork your own snapshot into the OpenHW organization, comment that and tag it. With that approach I strongly recommending deleting all save the master branch, updating the main `README.md` with a huge disclaimer at the top, and blocking any commits, other than rebasing on the upstream (i.e. this repo).

MikeOpenHWGroup, 2020-09-04:
Thanks for the quick reply @jeremybennett.   To your points:

> I've assigned to @neelgala who is one of the active maintainers.

Thanks!

> I note that so far the repository has not used tagging at all.

Yes, and I am starting to get the impression that few people in Git/GitHub/GitLab circles _do_ use them.   I find this odd as tags can be much more meaningful that hashes and are inexpensive.

> You might have to fork your own snapshot into the OpenHW organization, comment that and tag it. With that approach I strongly recommending deleting all save the master branch, updating the main [...]

Our strategy is to define a set of variables in our Makefiles for each repo we clone from.   These variables allow us to clone from a specific repo, branch, hash or tag.  Currently we've got:

```
COMPLIANCE_REPO   ?= https://github.com/riscv/riscv-compliance
COMPLIANCE_BRANCH ?= master
# 2020-08-19
COMPLIANCE_HASH   ?= c21a2e86afa3f7d4292a2dd26b759f3f29cde497
```
I would dearly love to replace that hash with a tag.

eroom1966, 2020-09-09:
@allenjbaum @neelgala 
Can somebody please action this request ?

allenjbaum, 2021-04-22:
There is now a tag that is (manually) updated on a merge, so closing this.

closed 2021-04-22 by allenjbaum