riscv/riscv-arch-test #137: The signature.outputs aren't identical with the references.output and report an error.

CODE-JTZ, 2020-10-13:
Hi. There is an error when I run the rv32i test suite on riscvOVPsim.  when it comes to the instruction JALR and MISALIGN_JMP, it reports that the signature.outputs aren't identical with the references.output. And I am still looking for the reasons that make it happen.
Hope for your reply, Thanks.
[image]

allenjbaum, 2020-10-13:
What core are you testing this on? IS this the entire diff?
This report appears to indicate that test (test A3, the first 2 parts  were
completely skipped, but the second 2 parts were not, nor were any following
tests - unless this isn't the entire diff.
This test assumes a core that can write the misa CSR and turn off the
C-extension, and that HW misalignment is not supported.
But, this reference signature indicates that the test has neither passed
nor failed - but that it wasn't executed at all (though the first 4 tests
were correctly executed)

jbwaite, 2020-10-14:
We have seen the same thing and wonder if the misa CSR be checked for writeable C-extension bit? And skip the test if not?

eroom1966, 2020-10-14:
The correct solution here is that the reference should be configured to match a (legal) representation of the target and produce a dynamic signature of the behaviour to compare

This means the reference must be able to model the correct RW behaviour of the MISA, and also the correct behaviour of MTVEC for when a trap occurs, this includes RW and trap vector alignment

allenjbaum, 2020-10-14:
This is correct; the particular reference in the test suite is set up for a
specific core configuration.
This will be fixed in the V.2 framework, which should be available near the
end of the year or beginning of next year.

V.2 of the framework will allow users to specify what their configuration
supports, and will both generate the reference signature dynamically to
match that configuration (using the SAIL formal model as the reference) and
will only select tests that are appropriate to the implementation (so, if
you don't implement M-extension, it won't try to test Mul/Div ops).
This will require implementers to describe their cores using a YAML
description; there is documentation in the riscv-config github repository,
and more documentation can be found here:

https://gitlab.com/incoresemi/riscof  (riscof  framework)
https://github.com/riscv/riscv-config/

The YAML description will enable you to specify which misa bits are RW,
which are read-only, and what their values are if read-only.

The v.2 framework will also bring a more capable trap handler, so the same
test could still run on a core that supports misalign load/store; it would
just give different results (and the reference signature would match).

allenjbaum, 2020-11-12:
@CODE-JTZ - can you please explain how you have configured riscvOVPsim, because we know that it has passed these tests previously?

CODE-JTZ, 2020-11-12:
> 
> 
> @CODE-JTZ - can you please explain how you have configured riscvOVPsim, because we know that it has passed these tests previously?

I just git clone the "riscv / riscv-compliance " and run the compliance tests by following the steps written in the README.txt file. The configuration of riscvOVPsim supposed to be in consist with the default way given in the codes. It appears that when I ignore the JALR and MISALIGN two instructions in rv32i, the other testsuites(rv32im rv32imc rv32Zicsr rv32Zifencei) 's verify can be right.

eroom1966, 2020-11-12:
Hi @CODE-JTZ 
I tried to follow your instructions

1. clone riscv-compliance

2. follow instructions in README.txt 
- I cannot find a file called README.txt ?
Do you mean README.md 

What is the make command line you are issuing ?
can you try the following

$ make clean
$ make RISCV_DEVICE=rv32i RISCV_ISA=rv32i VERBOSE=1 2>&1 | tee make.log

please post make.log here in this thread
Thx
Lee

CODE-JTZ, 2020-11-13:
> Hi @CODE-JTZ
> I tried to follow your instructions
>
>     1. clone riscv-compliance
>
>     2. follow instructions in README.txt
>
>
>     * I cannot find a file called README.txt ?
>       Do you mean [...]

CODE-JTZ, 2020-11-13:
Hi @eroom1966 
Sorry, It should be the README.md (not the *.txt).
And I follow your instructions.
[make.log](https://github.com/riscv/riscv-compliance/files/5534801/make.log)

allenjbaum, 2020-11-13:
I did a very quick look at JAL, and it appears that the signature dump
matches the reference, yet the test failed.

eroom1966, 2020-11-13:
@CODE-JTZ 
Its very strange, I wonder if this is somehow compiler related - can you describe what compiler you are using ?
Can you also supply the .elf file for the tests which miscompare ?
JALR and MISALIGN
Thx
Lee

CODE-JTZ, 2020-11-14:
> 
> 
> @CODE-JTZ
> Its very strange, I wonder if this is somehow compiler related - can you describe what compiler you are using ?
> Can you also supply the .elf file for the tests which miscompare ?
> JALR and MISALIGN
> Thx
> Lee
Indeed, the compiler prefix(RISCV_PREFIX ) I set in the top Makefile is "riscv64-linux-gnu-", which is different from the default value "riscv64-unknown-elf-". Could that be a problem? The other instructions in the rv32i (except JALR and MISALIGN_JMP) are just fine, they can produce the correct signature_outputs which can match the reference_outputs.
[elf.obj.zip](https://github.com/riscv/riscv-compliance/files/5540104/elf.obj.zip)

closed 2020-11-16 by CODE-JTZ

eroom1966, 2020-11-16:
OK, I see the issue.
You were using a linux compiler - not a bare metal compiler
I am unable to fully test this, as you did not provide the .elf files in the zip, you provided the objdump output
Thx
Lee

CODE-JTZ, 2020-11-16:
> 
> 
> OK, I see the issue.
> You were using a linux compiler - not a bare metal compiler
> I am unable to fully test this, as you did not provide the .elf files in the zip, you provided the objdump output
> Thx
> Lee

Sorry, my bad. Here are the two .elf files.
[file-elf.zip](https://github.com/riscv/riscv-compliance/files/5545909/file-elf.zip)

eroom1966, 2020-11-16:
I can now see the issue with your compilation
The compiler has inserted some instructions in a compressed format (2-byte)
this test is not intended to have compressed instruction support so it breaks 2 things 

1. your .elf is expecting support for the C extension
2. inclusion of C instructions has altered the alignment of instructions

thx
Lee

MarcKarasek, 2020-11-16:
Where do you see C extension opcode in the files?   I objdumped both elf
files and do not see any compressed opcodes.

I have attached both dump file..

eroom1966, 2020-11-16:
@MarcKarasek 
you are right I took a cursory look at the top of the dump and saw the following which is just the objdump program disassembling data.

> 00000000 <.note.gnu.build-id>:
>    0:	0004                	0x4
>    2:	0000                	unimp
>    4:	0014                	0x14
>    6:	0000                	unimp
>    8:	00000003          	lb	zero,0(zero) # 0 <_start-0x80000000>

So this will need further investigation to see why this is failing
Thx
Lee

neelgala, 2020-11-17:
I might have some lead here:

for I-MISALIGN_JMP-01 test the difference you are seeing in the signature dumps boils down [this case](https://github.com/riscv/riscv-compliance/blob/d51259b2a949be3af02e776c39e135402675ac9b/riscv-test-suite/rv32i/src/I-MISALIGN_JMP-01.S#L112-L116) failing. This case caues a misalign exception and should cause an entry to the ``_trap_handler``.

However, looking at the objdump of elf file the ``_trap_handler`` is loaded at address :

```
80000328 <_trap_handler>:                                                                           
```

But your mtvec seems to be initialized to the wrong value in the begining:

```
80000120: 00002097            auipc ra,0x2                                                          
80000124: 0f00a083            lw  ra,240(ra) # 80002210 <_GLOBAL_OFFSET_TABLE_+0xc>                 
80000128: 30509ff3            csrrw t6,mtvec,ra  
```
The content in ``80002210`` seems to be pointing to a GLOBAL_OFFSET_TABLE (GOT)

From the little that I know,  I understand this GOT is required when doing dynamic linking. If you are performing dynamic linking then you need to have a routine in the begining to load the programs/symbols into the locations.

my suggestion would be to use a bare-metal toolchain instead of the linux one.