riscv/riscv-arch-test #139: xori on 64 bit compiler needs neg number for 32 bit tests

MarcKarasek, 2020-10-28:
Compiling with 64 bit compiler does not see a 32 bit number as negative.  Need to implicitly set this as a negative number so
both 32 and 64 bit compilers behave properly.

From 54d5c11cde3a3675ea181a6ed2acd9e8e0633e92 Mon Sep 17 00:00:00 2001
From: Marc Karasek <mkarasek@inspiresemi.com>
Date: Wed, 28 Oct 2020 10:31:47 -0400
Subject: [PATCH 1/2] xori on 64 bit compiler needs neg number

Compiling with 64 bit compiler does not
see a 32 bit number as negative.  Need to
implicitly set this as a negative number so
both 32 and 64 bit compilers behave properly.
---
 riscv-test-suite/rv32i/src/I-RF_x0-01.S | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/riscv-test-suite/rv32i/src/I-RF_x0-01.S b/riscv-test-suite/rv32i/src/I-RF_x0-01.S
index 8f282d5..ddcc959 100644
--- a/riscv-test-suite/rv32i/src/I-RF_x0-01.S
+++ b/riscv-test-suite/rv32i/src/I-RF_x0-01.S
@@ -54,7 +54,7 @@ RV_COMPLIANCE_CODE_BEGIN
     addi    x0, x0, 1
     ori     x0, x0, 0x7F0
     andi    x0, x0, 0x53F
-    xori    x0, x0, 0xFFFFF803
+    xori    x0, x0, -0x7fd
     slli    x0, x0, 5
     srai    x0, x0, 2
     srli    x0, x0, 4
-- 
2.17.1

MarcKarasek, 2020-10-28:
[0001-xori-on-64-bit-compiler-needs-neg-number.patch.txt](https://github.com/riscv/riscv-compliance/files/5453724/0001-xori-on-64-bit-compiler-needs-neg-number.patch.txt)

allenjbaum, 2020-11-06:
The PR has been approved; can this be closed now?

MarcKarasek, 2020-11-06:
Yes i am a maintainer but in general one should not approve/close ones own
PR.  Just good SCM practice.

MarcKarasek, 2020-11-11:
This has been addressed in : https://github.com/riscv/riscv-compliance/pull/140

closed 2020-11-11 by MarcKarasek