riscv/riscv-arch-test #142: Not able to run compiance test for rv32E device and RV323E ISA.

subhajit26, 2020-11-18:
Hi,
    Is there any support to run the compliance test for RV32E as we need it as part of our thesis. I could successfully do it with RV32i since I can see they are already build in riscv-test-suite  directory but along with rv32i I also need it for rv32E.

subhajit26, 2020-11-18:
Hi,
     Can you please clarify the last part of your comment where you are mentioning that "it would be relatively easy to constrain the generator to meet specific coverage goals." Is it possible to generate the tests or is there any Makefile where I can make changes and build the target rules for RV32E  because  I have already build the riscv-gnu toolchain with RV32E arch and abi=ilp32e and was successful in compiling the C source file.

eroom1966, 2020-11-18:
@subhajit26 

> Can you please clarify the last part of your comment where you are mentioning that "it would be relatively easy to constrain the generator to meet specific coverage goals."

This is an assumption on my part, I think this is best answered by the architects of the generator @allenjbaum @neelgala 

> Is it possible to generate the tests or is there any Makefile where I can make changes and build the target rules for RV32E because I have already build the riscv-gnu toolchain with RV32E arch and abi=ilp32e and was successful in compiling the C source file.

I am not sure if the generator is publicly available (yet)
Also as I mentioned in my original email, the approach is Assembler based not C, in which case, having an RV32E C compiler does not help

neelgala, 2020-11-18:
@subhajit26 the test generator is indeed open source and available here: https://gitlab.com/incoresemi/riscv-compliance/riscv_ctg

As Lee mentioned it is indeed simple enough to declare new datasets and coverpoints using riscv_ctg for rv32e. You can simply re-use the samples/rv32i.cgf found in the repo and replace ``&all_regs`` with a dataset of x0-x15. 

I haven't tried this out yet and will be some time before I do. If you do attempt this please let me know what you find. You could also raise an issue in the riscv_ctg repo for better traction.

allenjbaum, 2020-11-18:
I'm sorry to say that we don't have RV32E tests at this time.
In fact, the RV32E spec hasn't been ratified yet; it's still in draft form.
You should be able to modify existing tests to use only registers x0..x15,
but it will be the work of a day or two.

Note that the compliance TG is not responsible for the existing riscv-test
directory.
Those are older tests and are unsupported by us.

allenjbaum, 2020-12-17:
WE have enabled tests for RV32E, but not RV32EM or RV32EC
If that isn't working for you, please respond with what is broken.
IF this works, but you need RV32EM or RV32EC, then we may not get it in. Let me know either way.

allenjbaum, 2021-04-26:
I'm going to close this next week unless I hear back from someone with an RV32E implementation that say that the current test won't work or you need RV32EC or RC32EM.

allenjbaum, 2021-08-17:
PR#199 fixes this, though the spec remains unratified

closed 2021-09-06 by allenjbaum